mongoose-os-apps/shelly-homekit #280: Shelly Button compatibility

DavRin, 2020-11-18:
Good morning, is it possible add the homekit compatibility on the shelly button (no homebridge)?

Thank you for your fantastic work

lucaspinelli85, 2020-11-18:
Inside the button there is a shelly1/1PM... Button is just a wrap...

lucaspinelli85, 2020-11-18:
Instead could be interesting the compatibility with the new BUTTON1 2020 (wireless instead wired). I don't know what there is inside (if the same shelly1 or something difference). @andyblac or @rojer do you know? And if it is already homekit compatible?
https://shelly.cloud/products/shelly-button-1-smart-home-automation-device/

andyblac, 2020-11-18:
and if your talking about Button1 add your support here https://github.com/mongoose-os-apps/shelly-homekit/issues/104

closed 2020-11-18 by andyblac

lucaspinelli85, 2020-11-18:
ok, so we can conclude that it is compatible with button (wired) and **not** compatible with button1 (wireless), right?
[image]

andyblac, 2020-11-18:
yes, button is just a shelly1.