mongoose-os-apps/shelly-homekit #281: Shelly Ad-on in Shelly 1 to trigger a boiler or connect to a Thermostat

Ludelag, 2020-11-18:
Dear all, there is any closer planning to add Shelly-Homekit the Add-on of shelly? 

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

Thank you.

[image]
[image]

lucaspinelli85, 2020-11-18:
you read my mind...

andyblac, 2020-11-18:
duplicate ticket https://github.com/mongoose-os-apps/shelly-homekit/issues/186 please add your support on that one.

closed 2020-11-18 by andyblac