mongoose-os-apps/shelly-homekit #282: Reported temperature too high in idle.

tggm, 2020-11-18:
Hello,

I just installed the latest version in roller shutter mode in my shelly 2.5 and everything works perfectly (thanks! 😄 ). However the web interface is showing 73º of temperature!

This value appears WAY to high. The device is idle and has been for a couple of hours. I used my multimeter thermometer to check around the box where the switch is installed and got  45º so it's probably  73ºC inside the switch (I noticed [issue 234](https://github.com/mongoose-os-apps/shelly-homekit/issues/234) so the values make sense)

Is this expected? The spec sheet of says the consumption is < 1W? I'm very worried about these temperature as there are live cables connected to the shelly and this may compromise the wire's insulation.

Is there a way to debug what's going on?

rojer, 2020-11-18:
well, what can i say - things get hot inside. ESP8266 itself heats up, and so does the voltage regulator, so PCB temperature is quite a bit higher than the ambient air temperature.
FWIW, when working on system temperature measurement, i verified readings by using multimeter thermocouple thermally connected directly to the NTC (using thermally conductive paste). readings produced by ESP8266 do not exactly match my multimeter, but were within 3C, which is good enough.
that said, mistakes happen. the way to debug it is to do what i did: bring a thermocouple in contact with the NTC inside shelly, and verify readings. use non-conductive thermally conductive paste to ensure proper contact (even tiny air gap will significantly impact readings).

tggm, 2020-11-20:
I'll check carefully the temperatures. It will take some days for me to do it. Btw, @rojer, what temperatures does your shelly 2.5 indicate when idle ?

rojer, 2020-11-21:
right now it's at 53C (ambient is 23)

tggm, 2020-11-21:
Well, I reverted to the stock firmware and temperatures dropped immediately to 62-64C and stable. Makes me wonder if there might be some loop in the code that's keeping the processor busy and causing these values.

rojer, 2020-11-22:
so, today i confirmed that temperature measurements on shelly 2.5 are ballpark accurate.
i initially developed support for it on shelly i3 and assumed that the circuit would be the same on all models - after all, why not?
well, now i actually tested it on shelly 2.5 with a thermocouple and heatgun - again, readings are a few degrees off, but no more than 5. this is good enough for the purpose.
i'm not sure what is the reason for such big difference with stock firmware measurement. the "busy loop" thing doesn't make much sense since CPU on shellies never sleeps anyway, i don't think stock fw uses power management either.
anyway, i'm perceiving this as a big issue. i confirmed temperature measurement to be a-ok, the rest is of academic interest, as far as i'm concerned.

closed 2020-11-22 by rojer

andyblac, 2020-11-22:
> anyway, i'm perceiving this as a big issue. i confirmed temperature measurement to be a-ok, the rest is of academic interest, as far as i'm concerned.

i presume you meant *not a big isssue ?

rojer, 2020-11-22:
oh, yes, you are right, i made a typo.

tggm, 2020-11-22:
For me it is a big issue. I consider these values unacceptable regardless if you see the problem as "academic" or not, so I'm moving away from this fw. This does not mean that the stock firmware temperatures (62-64C) are ok. But 75 and warm switches to the touch: thanks, but no thanks.

andyblac, 2020-11-22:
it must be location of shelly, my shelly's are in the wall cavity and i no issues with temps.
[image]

rojer, 2020-11-22:
it is not an issue because safety cutoff temperature if 100, 75 is way below that, so it's fine. differences between units and vs stock may be an interesting subject but i have way too much on my plate to pursue it.

jorgefrazaocosta, 2021-01-03:
Hello @rojer,
Just to confirm. For safety, Shelly will turn off at 100ºC? I was searching for this at the code.
It's supposed to be this line?  `sys_temp->reset(new TempSensorSDNT1608X103F3950(0, 3.3f, 33000.0f));`

rojer, 2021-01-03:
no, it's don in the [CheckOverheat() function](https://github.com/mongoose-os-apps/shelly-homekit/blob/3cfc021ac2736f52d6ed853bcad10fd93f77fd52/src/shelly_main.cpp#L450) on every status timer iteration (1 second).

jorgefrazaocosta, 2021-01-03:
Thanks. I see. 100ºC for turn off and when reach 90ºC it turn on again.