mongoose-os-apps/shelly-homekit #283: Shelly 1PM died due to possible bug.

Marfre888, 2020-11-19:
Hi @rojer 

I want to report a bug. I am not 100% sure this was the reason one of my Shelly 1PMs died, but it might be worth looking into it.

I have my shelly set to detached mode, with my Input 1 switch set to 'turn on = single, turn off= double'. A couple of days ago, my AppleTV was updating when a tried to turn on my lamp which has shelly attached to it. Ofc, nothing happened, as the AppleTV was off, so I flicked the switch a couple of times hoping it would jog it back to life before realizing that the reason it wasn't working was that my AppleTV was offline. A couple of seconds later the AppleTV went online, and all these instructions to turn on/off went on at the same time, causing the lamp to turn on/off for a couple of seconds without me touching the switch. This also caused the shelly to start to spark rapidly and die.

Shouldn't there be a form of timeout where if an instruction isn't executed within a couple of ms it is lost?

rojer, 2020-12-13:
> This also caused the shelly to start to spark rapidly and die.

with all the activity you probably triggered wifi reset - if within the first minute of boot there are several transitions on input 1, shelly resets wifi and goes into AP mode. this is one of the [recovery methods](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery).

closed 2020-12-13 by rojer

Marfre888, 2020-12-13:
@rojer Shelly device was on for 8days when AppleTV was updating which killed detached mode functionality.

Either way, if it did or did not trigger a wifi reset it fried the shelly and it won’t turn back on.

Please for safety purposes, could you take a look at the code and check if it’s possible to a series of commands to trigger at once, frying the shelly

Thanks