mongoose-os-apps/shelly-homekit #285: flash_shelly.py error

geoigeek, 2020-11-20:
Hello,

i want to use the Flash Script for the first time in my MBP 11.0.1

But i get the following Error. What does it mean?

```
./flash_shelly.py -la
Scanning for Shelly devices...

Traceback (most recent call last):
  File "/Users/alex/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 539, in <module>
    device_scan(args.hosts, action, args.do_all, args.dry_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, stock_release_info, homekit_release_info)
  File "/Users/alex/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 469, in device_scan
    parse_info(device, action, dry_run, silent_run, mode, exclude, version, variant, stock_release_info, homekit_release_info)
  File "/Users/alex/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 312, in parse_info
    stock_model = device_info['stock_model']
KeyError: 'stock_model'
```
Thx

andyblac, 2020-11-20:
please rerun with `./flash_shelly.py -la -v=1` and post results, thx

geoigeek, 2020-11-20:
here are the results

```
./flash_shelly.py -la -v=1
app
OS: Darwin
action: list
mode: homekit
do_all: True
dry_run: False
silent_run: False
mode: homekit
version: False
exclude: None
variant: False
verbose: 1
hosts: []

stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://api.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20200827-065544/v1.8.3@4a8bc427'}, 'SHPLG-S': {'url': 'http://api.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20200827-070306/v1.8.3@4a8bc427'}, 'SHPLG-U1': {'url': 'http://api.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20200827-070340/v1.8.3@4a8bc427'}, 'SHPLG2-1': {'url': 'http://api.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20200827-070415/v1.8.3@4a8bc427'}, 'SHSW-1': {'url': 'http://api.shelly.cloud/firmware/SHSW-1.zip', 'version': '20200827-065344/v1.8.3@4a8bc427'}, 'SHSW-21': {'url': 'http://api.shelly.cloud/firmware/SHSW-21.zip', 'version': '20200827-065420/v1.8.3@4a8bc427'}, 'SHSW-25': {'url': 'http://api.shelly.cloud/firmware/SHSW-25.zip', 'version': '20200827-065456/v1.8.3@4a8bc427'}, 'SHSW-PM': {'url': 'http://api.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20200827-070450/v1.8.3@4a8bc427'}, 'SHAIR-1': {'url': 'http://api.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20200812-092044/v1.8.0@8acf41b0'}, 'SHSW-44': {'url': 'http://api.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://api.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20200826-111450/1206-production-ShellyUNI@a80ee1fc+'}, 'SHEM': {'url': 'http://api.shelly.cloud/firmware/SHEM.zip', 'version': '20200813-140346/v1.8.1@1b2a49be'}, 'SHEM-3': {'url': 'http://api.shelly.cloud/firmware/SHEM-3.zip', 'version': '20200813-140420/v1.8.1@1b2a49be'}, 'SHSEN-1': {'url': 'http://api.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHGS-1': {'url': 'http://api.shelly.cloud/firmware/SHGS-1.zip', 'version': '20200812-092502/v1.8.0@8acf41b0'}, 'SHSM-01': {'url': 'http://api.shelly.cloud/firmware/SHSM-01.zip', 'version': '20200812-091144/v1.8.0@8acf41b0'}, 'SHHT-1': {'url': 'http://api.shelly.cloud/firmware/SHHT-1.zip', 'version': '20200812-091311/v1.8.0@8acf41b0'}, 'SHWT-1': {'url': 'http://api.shelly.cloud/firmware/SHWT-1.zip', 'version': '20200812-091355/v1.8.0@8acf41b0'}, 'SHDW-1': {'url': 'http://api.shelly.cloud/firmware/SHDW-1.zip', 'version': '20200818-120325/v1.8.2@36539b0b', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHDW-1.zip', 'beta_ver': '20200814-174756/v1.8.2-rc1@59a25feb'}, 'SHDW-2': {'url': 'http://api.shelly.cloud/firmware/SHDW-2.zip', 'version': '20200818-120407/v1.8.2@36539b0b', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHDW-2.zip', 'beta_ver': '20200814-174839/v1.8.2-rc1@59a25feb'}, 'SHSPOT-1': {'url': 'http://api.shelly.cloud/firmware/SHSPOT-1.zip', 'version': '20200812-090754/v1.8.0@8acf41b0'}, 'SHCL-255': {'url': 'http://api.shelly.cloud/firmware/compatibility/SHCL-255/0.2.0-73a40eef.zip', 'version': '20180116-161402/v0.2.0@73a40eef'}, 'SHBLB-1': {'url': 'http://api.shelly.cloud/firmware/SHBLB-1.zip', 'version': '20200812-090503/v1.8.0@8acf41b0'}, 'SHCB-1': {'url': 'http://api.shelly.cloud/firmware/SHCB-1.zip', 'version': '20200812-090429/v1.8.0@8acf41b0'}, 'SHRGBW2': {'url': 'http://api.shelly.cloud/firmware/SHRGBW2.zip', 'version': '20200827-070122/v1.8.3@4a8bc427'}, 'SHRGBWW-01': {'url': 'http://api.shelly.cloud/firmware/SHRGBWW-01.zip', 'version': '20200827-065125/v1.8.3@4a8bc427'}, 'SH2LED-1': {'url': 'http://api.shelly.cloud/firmware/SH2LED-1.zip', 'version': '20200812-090719/v1.8.0@8acf41b0'}, 'SHDM-1': {'url': 'http://api.shelly.cloud/firmware/SHDM-1.zip', 'version': '20201007-085347/v1.8.5@5fbcbaa3'}, 'SHDM-2': {'url': 'http://api.shelly.cloud/firmware/SHDM-2.zip', 'version': '20201007-085431/v1.8.5@5fbcbaa3'}, 'SHDIMW-1': {'url': 'http://api.shelly.cloud/firmware/SHDIMW-1.zip', 'version': '20200812-092244/v1.8.0@8acf41b0'}, 'SHVIN-1': {'url': 'http://api.shelly.cloud/firmware/SHVIN-1.zip', 'version': '20200827-065016/v1.8.3@4a8bc427'}, 'SHBDUO-1': {'url': 'http://api.shelly.cloud/firmware/SHBDUO-1.zip', 'version': '20200827-065050/v1.8.3@4a8bc427'}, 'SHBTN-1': {'url': 'http://api.shelly.cloud/firmware/SHBTN-1.zip', 'version': '20200827-070047/v1.8.3@4a8bc427'}, 'SHIX3-1': {'url': 'http://api.shelly.cloud/firmware/SHIX3-1.zip', 'version': '20200827-070910/v1.8.3@4a8bc427'}}}

homekit_release_info:[['X^2.5.0-.*', {'version': '2.5.0-beta2', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/issues/254#issuecomment-723514784', 'rel_notesX': '', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-Shelly1.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-Shelly1PM.zip', 'Shelly2': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-Shelly2.zip', 'Shelly25': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-Shelly25.zip', 'ShellyI3': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-ShellyI3.zip', 'ShellyPlug': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-ShellyPlug.zip', 'ShellyPlugS': 'https://rojer.me/files/shelly/2.5.0-beta2/shelly-homekit-ShellyPlugS.zip'}}], ['.*', {'version': '2.5.0', 'rel_notes': 'https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.5.0', 'urls': {'Shelly1': 'https://rojer.me/files/shelly/2.5.0/shelly-homekit-Shelly1.zip', 'Shelly1PM': 'https://rojer.me/files/shelly/2.5.0/shelly-homekit-Shelly1PM.zip', [... cut ...]
    device_scan(args.hosts, action, args.do_all, args.dry_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, stock_release_info, homekit_release_info)
  File "/Users/alex/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 469, in device_scan
    parse_info(device, action, dry_run, silent_run, mode, exclude, version, variant, stock_release_info, homekit_release_info)
  File "/Users/alex/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 312, in parse_info
    stock_model = device_info['stock_model']
KeyError: 'stock_model'
```

andyblac, 2020-11-20:
ok, your on a fairly old version (2.0.12) this version did not include the required infomation for the script, please update via the webUI, then the script should work. i'll add in a version check to stop crash thanks.

PR #286 (2020-11-21): [Flashscript] add version check, add some info.
fixes #285
closed 2020-11-21 by rojer