mongoose-os-apps/shelly-homekit #287: Firmware does not change??

j55marshall, 2020-11-22:
Hi guys the instructions look so simple,  however I have tried everything including installing Mongoose HTTP server and no difference.

My browser goes to a page like this:

{"status":"updating","has_update":false,"new_version":"","old_version":"20190329-090914/master@b49f177f"}

Nothing seems to happen,  I have a steady ping hitting the device and it never reboots just sits on stock firmware?
Thanks

andyblac, 2020-11-22:
update stock firmware 1st, that too old. then retry.

j55marshall, 2020-11-22:
Ok thanks but when I check it says that it is update to date see attached
[image]

rojer, 2020-11-22:
interesting. what device is this?

j55marshall, 2020-11-22:
Shelly1 PM

j55marshall, 2020-11-22:
Ok got a bit further here.   The device was attached to one of my TINY pcs WIFI direct.   I have joined to my actual main WIFI network and then I did get an option to upgrade the FW.

Now it appears I am on the latest FW,  you can see many more stock config options.

I then went to the OTA link and this is what I get:

LINK
http://192.168.0.70/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1PM.zip

PAGE:
{"status":"updating","has_update":false,"new_version":"20200827-070450/v1.8.3@4a8bc427","old_version":"20200827-070450/v1.8.3@4a8bc427"}

Still on the stock FW :/. so close!!  :)

j55marshall, 2020-11-22:
WOW got it!

[image]

j55marshall, 2020-11-22:
I guess had to be more patient for it to take thanks everyone

closed 2020-11-22 by j55marshall

andyblac, 2020-11-22:
glad you got it sorted, flashing can take 30 seconds