mongoose-os-apps/shelly-homekit #288: Shelly going into AP mode all by itself

pivale, 2020-11-24:
Hi,

I've noticed a strange behaviour. From time to time, can't really say how frequent it is, one shelly will go into AP mode.
I then need to connect directly and only switch wifi on again, and all good!

Notes:
- This has happened at least once with shelly 1pm, but I think it happened already with one shelly 2.5
- Last time, uptime was 7d when I noticed, so it doesn't seem to be related to uptime (my other shellys have similar uptime and all good)
- This seems to be a rare event. I have shellys for a couple of months and it happened 3 or 4 times in different ones.
- I couldn't see anything in the logs, since when I noticed that it was in AP mode it was probably long gone (hint, this would be helpful in situations like this ;) #236)
- The physical switch kept working as normal (as webui and everything else except not being connected to wifi)
- The shellys are in a separate wifi network, all in the same one, with 2 APs and none of the other shellys connected to the same AP had the same problem (neither are they the ones with weaker signal)

Not sure how to help here, I don't think this is enough information or frequent enough, but maybe someone has the same problem and can add more info

rojer, 2020-11-24:
i have indeed observed this in 1PM in particular - looks like its button input is noisy.
but this was supposed to be fixed in 2.5.0, is your device running 2.5.0?

rojer, 2020-11-24:
this was the fix - https://github.com/mongoose-os-apps/shelly-homekit/commit/8898bce2dc7757eb4e2692fb1130682f6fc50fa2

pivale, 2020-11-24:
Yes, I have all my devices in 2.5.0.

rojer, 2020-11-24:
hm, this is really strange but i guess it can happen. i'll need to improve noise suppression further.

rojer, 2020-11-24:
@pivale does this happen with any regularity? just wondering how to test possible fixes.

pivale, 2020-11-24:
I couldn't really tell... 
I have shellys since early September and I saw this happening 3 times or so... One of them I think it was in a Shelly 2.5.

Btw, not sure if related or not, but my other 1pm lost Wi-Fi, didn't went in AP mode or saw any blinking led... (physical switch still worked). 
I had to disconnect it from main and reconnect and it came back to life.

rojer, 2020-11-25:
try 2.6.0-beta - https://github.com/mongoose-os-apps/shelly-homekit/issues/291

rojer, 2020-12-13:
i think this should be fixed, please reopen if not.

closed 2020-12-13 by rojer