mongoose-os-apps/shelly-homekit #289: Notifications are they posible ?

andyblac, 2020-11-24:
would it be posible to add the ability to send a notification if switch ON or OFF for a set period of time ?, this could be usefull for Garage door for example, IF door is open for xx mins send a Notification, As HomeKit can't do this natively via automations.

rojer, 2020-11-24:
we don't control notifications sent to the user. all we can do is notify of state changes, i.e. open to closing.
i guess after N minutes in open state we could simulate closing and then immediately back to open, which should generate a new "door is open" notification.

andyblac, 2020-11-24:
what about 'light is on' etc, so HAP does not have access to notifications ?

rojer, 2020-11-24:
no, these notifications are generated by the Home app in response to change of the "on" characteristic.

andyblac, 2020-11-24:
so, what are the ways we can 'trigger' them ? for you left a light on for example

rojer, 2020-11-24:
turn it off and back on. but this will generate "light is off" too.

andyblac, 2020-11-24:
but i do not see 'Notication" opion on lights, also if this is possiible, could this be 'simulated' so it dose not acutaly control the light ?

rojer, 2020-11-24:
ok, let me put it this way: device has no direct control over notifications sent to the user, controller makes that decision.
in practice i see user-level notifications generated in response to chnages to state of lock (locked/unlocked) and garage door (open/closed).

andyblac, 2020-11-24:
ye, i get that, but i don't see anyway to send one for "you left bathroom light on", so i guess this is not possible ?

rojer, 2020-11-24:
probably not for light. specifically for garage door it may be possible because home app does generate user notifications for it.

andyblac, 2020-11-24:
> probably not for light. specifically for garage door it may be possible because home app does generate user notifications for it.

ok thought so, i'll leave this ticket open, as it would have some usefulness.

andyblac, 2020-11-24:
IMO, this is something that Apple should add to Home app, if you have any contacts at Apple maybe you could ask ?

rojer, 2020-11-24:
no, i do not

jsiegenthaler, 2020-11-24:
I think it is an Apple issue too. Stick to the protocol (HAP) and you won't go wrong. Start deviating from it and... who knows what may happen...

closed 2020-12-02 by rojer