mongoose-os-apps/shelly-homekit #290: Request: Inverted (Reversed) SW output

lucaspinelli85, 2020-11-25:
#220

lucaspinelli85, 2020-12-03:
Like the inverted inputs I need the inverted outputs

[image]

lucaspinelli85, 2020-12-03:
Principally difference from inverted inputs is in the app, that show me inverted state of relay rather show me inverted state of switch when relay and switch are inverted!

rojer, 2020-12-03:
this one i'm not so sure about. give me a use case for this.

rojer, 2020-12-13:
um... no, i don't like the idea after all.

closed 2020-12-13 by rojer

tvarohp, 2024-02-24:
I see you have talked about invert ouputs, but I do not see solution. Do you have any soulution for Shelly 1 and shelly PM for THAT?
Use case: I have No and Nc valves. I use only buttons in apps.(I have no switches on inputs connected) I need to unify buttons to show ( on= all NC and NO are OPEN, and wise versa) could you suggest me the solution for that?