mongoose-os-apps/shelly-homekit #293: Manual Calibration 2.5

busaku, 2020-11-27:
Hey, 

It is possible to set up the runtime of shelly2.5 manually?

My shelly is a "master" jalousie. So it means, it trigger 5 other roller-shutter. An that mean, there is no power-consumption and no calibration possible.

Thank you in advance.

Patrick

rojer, 2020-11-27:
while there is a way to manually set the "calibrated" flag and the movement time, it won't really help as power consumption is required during operation.
maybe if you describe your setup in more detail we'll be able to figure something out.

busaku, 2020-11-27:
oh, okay, why is power consumption is needed?

so the way is:
```
power - shelly - relay - engine
     |----------------|
```

that means, the shelly just triggert the relay in front of the engine. and the engine put the power through the engine.
so, the shelly only needs a few mw to trigger the relay.

my english is not the yellow from the egg ;) 
so i hope you understand what i mean.

thanks

rojer, 2020-11-27:
yes, i understand... unfortunately, right now the roller shutter mode expects to directly drive the curtains, and it monitors power consumption to tell if movement is still going on or not.

in your system, sound like you need a simple switch.

rojer, 2020-11-27:
i mean - shelly 2.5 in switch mode, maybe with auto off, should be fine for your setup.

lucaspinelli85, 2020-11-27:
> oh, okay, why is power consumption is needed?

I think power is needed to determinate position. what you ask is to calibrate by time instead of power? You can just put the timer auto off

busaku, 2020-11-27:
@lucaspinelli85 ah thank you. you mean the same as @rojer ?
set it up as switch and only setup the time?

lucaspinelli85, 2020-11-27:
> @lucaspinelli85 ah thank you. you mean the same as @rojer ?

Yes, I'm just a few seconds late ..

busaku, 2020-11-27:
thank you both for response.

i will ask the electrician if its possible to change the wires. 
i like to use the percentage  of current opening ;)

closed 2020-11-27 by busaku

lucaspinelli85, 2020-11-27:
> set it up as switch and only setup the time?

Yes, but, i don't know your application with motor, but you loose the possibility to place the stroke where you want like a roller schutter, that's the purpose of schelly25

torto-85, 2021-01-08:
> Hey,
> 
> It is possible to set up the runtime of shelly2.5 manually?
> 
> My shelly is a "master" jalousie. So it means, it trigger 5 other roller-shutter. An that mean, there is no power-consumption and no calibration possible.
> 
> Thank you in advance.
> 
> Patrick

Hi Patrick, Rojer...

first of all: THANKS FOR THE AMAZING FIRMWARE!!

I have the same case like Patrick. And it works perfect with "switch mode". But can I setup that both buttons (switches) don't work at the same time to protect the motor against destruction?

I would be glad if you could help!
Thanks, Torto