mongoose-os-apps/shelly-homekit #294: Shelly as Motion Sensor in HomeKit

jobl1978, 2020-11-27:
I have a wired motion sensor from Berker that have 2 modes

Mode 1:
You can set how long the switch is closed after the last motion (10sec to x-sec)
The sensor controls the timeout and turns of the output

Mode 2:
The motion sensor can act with relays, in this case the motion sensor has no timeout and will only send a 200ms signal after motion detection and doesn't keep close the output. If a signal is triggered the motion sensor stops detection for motions for 10sec. After 10sec it starts again detecting motion and fires a new 200ms signal to the output and so on.

HomeKit Motion Sensor behavior

Mode 1:
Initial state -> state of sensor output
As long the output is closed it is in state "Motion detected" otherwise it is in "No Motion" state

Mode 2:
Initial state -> no motion
200ms signal received -> Motion detected state
If a motion was detected and the signal doesn't come again within an configurable delay -> No Motion state

What do you say about that? I think most of wired motion sensors will work in one or both of my described ways. Would be a big improvement if this will open the world for high quality motion sensors to HomeKit. Hue and Eve sucks because they are battery driven and doesn't fit to the design of existing switch programs like Berker, Gira, Busch Jäger etc.

rojer, 2020-11-27:
yes, i think this makes sense.
i see it as a configuration option for the detached input.
right now, when you put inputs into detached state, it is exported to homekit as a stateless switch.
we should add a setting to instead export it as a motion sensor, with additional option to select for level and pulse mode (1 and 2 described above, respectively).

Cojaeck, 2020-11-28:
This would really be great. I use a lot of the devices in detached mode. What also would be great is the option to let it appear as a occupancy sensor. Then it would satisfy all my needs :-)

Cojaeck, 2020-11-28:
I already used this with my shelly devices in homekit via homebridge and webhooks. To have this in the shelly-homekit-firmware would really be a killer feature for me

rojer, 2020-12-03:
Ok, here's an alpha version with motion and occupancy sensor modes: http://rojer.me/files/shelly/2.7.0-alpha1/
In addition to stateless switch, detached inputs can be either motion or occupancy sensors. Both level and pulse modes are supported, as requested. @jobl1978 please test and let me know.

Another mode for an input is "disabled", this is to address https://github.com/mongoose-os-apps/shelly-homekit/issues/318 - @timoschilling, please test

Also in this release: more accurate system temperature measurement - @b2a3e8, you should see lower temperatures for your i3 devices.

jobl1978, 2020-12-04:
Hi @rojer, thank your! I tested the new firmware. I configured the input as detached/motion sensor in pulse mode with 15sec idle time. The motion will be detected well. I also configured the auto-off in the web ui to 1min.

Problems I saw:

- If motion is detected and my motion sensor triggers a further pulse within the 15sec the pulse mode idle timer will not be reseted and the state always changes to "no motion" after 15sec
- In Home App the light is still shown as on after the firmware auto-off timer turns the light off

Cojaeck, 2020-12-04:
Hi @rojer,

Thank you very much.
I also tested the alpha release and it does exactly what I want it to do.
I have one input of a shelly 2.5 set to Occupancy, the other to motion sensor.
Both in level mode.
In HomeKit they appear as expected and react as expected.
Great work.
Now there is no need for HomeBride anymore for me.
Your great firmware does all I want :-)

Kind Regards
Michael

rojer, 2020-12-04:
@jobl1978 fixed in alpha2, please update

jobl1978, 2020-12-04:
> @jobl1978 fixed in alpha2, please update

Tested with Alpha2 and works now as expected. Thx @rojer

jobl1978, 2020-12-04:
@rojer Do have any idea, why the light state is not updated in the Home App if the state changes? I always have to restart the Home App to get the correct state shown for that specific Shelly that is running with the new Firmware as motion sensor. If I activate/deactivate the light with the Home App button, the state changes. But if an automation or Shellys input switch turns on the light, the state does not change in Home App. Is there an event firing missing in the firmware when the state changes?

rojer, 2020-12-04:
this looks like we may not be sending a notification when auto-off timer fires. i'll verify.
to clarify: you have a shelly in switch mode with a light connected and auto-off configured.
is it turned on from home app via automation based on the motion sensor?

jobl1978, 2020-12-04:
> this looks like we may not be sending a notification when auto-off timer fires. i'll verify.
> to clarify: you have a shelly in switch mode with a light connected and auto-off configured.
> is it turned on from home app via automation based on the motion sensor?

The light itself (hardware) is always turning off/on correctly but the state in the Home App does not reflecting this anymore without restarting the Home App. This is not an auto-off only issue. I also deactivated my Apple TV some days ago since I bought a HomePod (not mini) as HomeKit control device because Apple TVs are running crazy as @andyblac already mentioned.

[image]

jobl1978, 2020-12-04:
I also found out now, that moving Home App to background and then back to foreground also updates the state correctly. Restarting the app is not needed.

andyblac, 2020-12-04:
this is a Home app issue, I get same issue with official HomeKit devices. it seems status can some times get stuck.

jobl1978, 2020-12-04:
> this is a Home app issue, I get same issue with official HomeKit devices. it seems status can some times get stuck.

No, I don't think that this is a HomeKit issue. I updated now a second Shelly1 with the new firmware (runs as momentary switch) and now this Shelly does also not reflecting the state anymore without restarting the App. This Shelly worked before. Every other Shelly with 2.6.1 still works.

jobl1978, 2020-12-04:
I did another test to verify this. I switched this second Shelly1 back to 2.6.1 than state changes are working again.

jobl1978, 2020-12-04:
Summary for me it looks like that the state is correctly set inside the Shelly firmware. But state changes are not reported actively from the firmware to HomeKit. The Home App forces state queries only at app startup or if app switches from background to foreground. If the Home App is already started and in foreground it only queries the state if a event comes in which doesn't come.

jobl1978, 2020-12-04:
I compiled a fresh firmware for Shelly1 with an additional logging to verify if "RaiseEvent" is called and the state caching is working correct. RaiseEvent is called, but it does not update state in HomeApp. Im not sure if the call 'HAPAccessoryServerRaiseEvent' inside "RaiseEvent" is reached. Didn't found a way to log from mgos sources.
[image]

rojer, 2020-12-04:
@jobl1978 found the issue. it was a regression in how HAP services are registered, introduced during refactoring related to adding motion sensor. https://github.com/mongoose-os-apps/shelly-homekit/commit/3c9cb2cc8a2e47f0aa9fdf1b89513c2574919aa5 is the fix. built and pushed alpha3, please update and test.

jobl1978, 2020-12-04:
@rojer Oh yeah! I works now!!! Thx

andyblac, 2020-12-05:
@rojer in alpha 6 the webUI ? icon is pointing to 2 diferant url's 

Disabled / Stateless Switch: it is pointing to `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings#detached-input
Motion Sensor / Occupancy Sensor / Contact Sensor: it is pointing to `https://github.com/mongoose-os-apps/shelly-homekit/wiki/MotionSensor`

can you change them to point to:
Disabled: `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#disabled`
Stateless Switch: `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#stateless-switch`
Motion Sensor: `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#motion-sensor`
Occupancy Sensor: `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#occupancy-sensor`
Contact Sensor: `https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#contact-sensor`

and I'll create new page for it.

rojer, 2020-12-05:
@andyblac good point. since all the sensors share the same template, i'd need to tinker with the url in javascript to modify it.
since sensor modes are all more or less the same, i think it's fine to have one section that describes them all.
go ahead, create the page and i'll use URLs from it to update HTML.

jobl1978, 2020-12-07:
@rojer I saw that you did some OOP reworks to sensors. I tested the motion sensor in pulse mode again with 2.7.0-alpha6. Everything still works as expected. The only thing I saw, after I updated the firmware from the web-ui, some settings where reseted and I had to configure the motion sensor type again to pulse.

timoschilling, 2020-12-09:
@rojer the disable feature works well, thanks!

timoschilling, 2020-12-09:
I updated my Shelly i3 with 2.7.0-alpha6 after the installation it was web ui was reachable, but after a minute it becomes unavailable and didn't reconnect via wifi. Event turn power off and on didn't bring it back online.

rojer, 2020-12-13:
please give 2.7.0 beta a go.
i'm going to close this issue, as functionality has been implemented and will be released after more testing. please use #330 to report any additional issues you find.

closed 2020-12-13 by rojer