mongoose-os-apps/shelly-homekit #295: Shelly Flood

obudik, 2020-11-27:
Will it be possible to connect HomeKit to Shelly Flood?

Saronovic, 2020-11-27:
And maybe the Shelly H&T would be nice. For example during summertime you can close your roller shutter when it is getting hotter in your house.

miamilabs, 2020-11-28:
And Shelly Gas =)

andyblac, 2020-11-29:
> And maybe the Shelly H&T would be nice. For example during summertime you can close your roller shutter when it is getting hotter in your house.

a ticket is already created here https://github.com/mongoose-os-apps/shelly-homekit/issues/186 those.

rojer, 2022-06-29:
Flood is battery-operated and i think is based on a different chip, so it's in a very distant future, if ever.

timoschilling, 2022-06-29:
> based on a different chip

Nope its a ESP8266, but maybe with a co-processor.

vrpt2022, 2022-06-29:
> Flood is battery-operated and i think is based on a different chip, so it's in a very distant future, if ever.

Ok. Thanks for your answer.

By this answer I assume that mongoose os version for the battery operated devices (shelly flood, H&T and Door/Window 2 sensor) won't we available any soon, right?

It would be grate to have them working with mongoose OS, other brand sensor are 2-3 times more expensive or need external app/hubs to work.

Again, thanks for the great work all of you have done until now.

timoschilling, 2022-07-03:
The problem is that WiFi isn’t made for battery based iot devices. All HomeKit battery devices I know are using Bluetooth, ZigBee or other low energy standards. That’s why we can’t support that device natively, maybe we will add some kind of bridge support to wired Shelly so that you don’t need Bridges.

timoschilling, 2023-02-21:
replaced by https://github.com/mongoose-os-apps/shelly-homekit/issues/1233

closed 2023-02-21 by timoschilling