mongoose-os-apps/shelly-homekit #297: Shelly i3 like a sensor status view

lucaspinelli85, 2020-11-28:
Hi all.
I am integrating shellyes into the alarm control panel.
Arm/Disarm performed by shelly1.
Now i want to see in app Home the status of various sensors (windows, doors...) if are open or closed (only view! because there is no control of that! I can't close the window remotely..).
So I chosen shelly i3 because I don't need an output (relay) but only see the input status and I can see 3 toggle in unique device.
The problem is that in the app I see the toggle like a bridge. I don't need this because I don't have to command anything, I just want to see if the contact is open or closed!
There is an "switch mode" to do this? it's the right way to do what I'm looking for?
There is something similar in stock firmware?
I hope you understand my purpose, if you have other better solutions I'm happy!

rojer, 2020-11-28:
so you want stateful status of i3.
i think that would be great as well, but i don't see a suitable standard HAP service to use.
perhaps Contact Sensor, Occupancy Sensor or Motion Sensor

jsiegenthaler, 2020-11-28:
I think a sensor would be the best but let the user configure what sensor he wants. I use fake motion sensors and occupancy sensors to track people at home. Different sensors behave differently, so let the user have the choice.

jsiegenthaler, 2020-11-29:
Actually I think an i3 would best map to a Homekit StatelessSwitch

rojer, 2020-11-29:
that's how it's represented today

Dabr422, 2020-12-05:
I have a similar application in mind, where I plan to use the Shelly i3 to read status signals from my burglar alarm. There are three signals that I want to read and expose to Homekit - "Fire alarm", "Burglar alarm" and "alarm system enabled". When a fire or burglar alarm is triggered I would like to get a notification in my phone. I've therefore bought an i3 but now realize that it doesn't behave as I need it to in order to acheive what I want to do. I would like it to be configured as a contact sensor.
A bonus feature would be if it could be set a filter time before a new notification is triggered since my fire alarm output is pulsating with a frequency of something like 0,5-1Hz. So if triggered it will be a lot of notifications otherwise :-). But if there is a fire in my house I have bigger problems than getting a lot of notifications so... just a bonus.
Any chance this can be realized??

rojer, 2020-12-05:
ok, so you want a contact sensor representation. that can be done, i've just added motion and occupancy sensor modes, contact sensor is pretty much the same

rojer, 2020-12-05:
added contact sensor mode to alpha6, please test - https://github.com/mongoose-os-apps/shelly-homekit/issues/294#issuecomment-738405359

rojer, 2020-12-05:
> A bonus feature would be if it could be set a filter time before a new notification is triggered since my fire alarm output is pulsating with a frequency of something like 0,5-1Hz. So if triggered it will be a lot of notifications otherwise :-)

you should be fine if you put sensor into pulse mode: in pulse mode, it activates when there's a 0 -> 1 transition and stays on until there are no pulses for idle_time period (15 sec by default)

Dabr422, 2020-12-06:
Wow, didn't expect a fix so soon! Thanks :-). I've spent today connecting my Shellys to the alarm system and finding the best configuration in the Shellys and alarm system sides. Works very well now. I ended up using the motion sensor alternative since the notification text then made most sense. It would have been great to be able to control the notification text but I realize that this is handled within Homekit.

rojer, 2020-12-13:
that's right, device does not control notification text.

ok, i'm going to close this issue, feature is implemented and will be generally available in 2.7.0 release.
you can help test pre-releases here - https://github.com/mongoose-os-apps/shelly-homekit/issues/330

closed 2020-12-13 by rojer