mongoose-os-apps/shelly-homekit #298: Need help flash link dienst work

acap86, 2020-11-28:
Hi,

i know that here are many Issues closed already with a similar subject. I hope you have the patience to help a newbie again with this problem.

the link does not work for me:
http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip

what i did already:
-i replaced the abcd with my shelly ip
-i am shure that my shelly has internet connection because i thought that firmware 1.9 could be too new
- the downgrade link from shelly work fine
- after flashing with your link i waited over 10 min to go to my shelly 2.5

i am really happy about your work and it would be great if i could use your FW.

Thx and best regards!
Andy

flashfabrixx, 2020-11-28:
Hi there! Currently experiencing the same problem, even when using the script from the terminal. 

Had the following error message while trying to flash the device: 

```
flashfabrixx@Marcel-MBP tools % ./flash_shelly.py shellyswitch25-40F5202A2C27.local
Probing Shelly device for info...

Host: shellyswitch25-40F5202A2C27.local
IP: 192.168.178.88
Model: Shelly25
Current: Official 1.9.0
Latest: HomeKit 2.6.0
Do you wish to flash shellyswitch25-40F5202A2C27 to firmware version 2.6.0 (y/n) ? y
Now Flashing...
waiting for shellyswitch25-40F5202A2C27 to reboot...
still waiting for shellyswitch25-40F5202A2C27 to reboot...
Traceback (most recent call last):
  File "/Users/flashfabrixx/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 574, in <module>
    device_scan(args.hosts, action, args.do_all, args.dry_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, stock_release_info, homekit_release_info)
  File "/Users/flashfabrixx/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 503, in device_scan
    parse_info(device, action, dry_run, silent_run, mode, exclude, version, variant, stock_release_info, homekit_release_info)
  File "/Users/flashfabrixx/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 466, in parse_info
    write_flash(host, lfw, dlurl, cfw_type, mode, requires_upgrade)
  File "/Users/flashfabrixx/Downloads/shelly-homekit-master/tools/./flash_shelly.py", line 301, in write_flash
    if info['stock_model'] == 'SHRGBW2':
TypeError: 'NoneType' object is not subscriptable
```

istefanpop, 2020-11-28:
Having the same issue with 2.5...  Revert seems to work from 1.9.0 to 1.9.2... 

Host: shellyswitch25-BCDDC27768EF.local
IP: 192.168.1.100
Model: Shelly25
Current: Official 1.9.2
Latest: HomeKit 2.6.0
D_URL: https://rojer.me/files/shelly/2.6.0/shelly-homekit-Shelly25.zip
perform_flash: True
Do you wish to flash shellyswitch25-BCDDC27768EF to firmware version 2.6.0 (y/n) ? y

write_flash
Now Flashing...
curl -qsS http://shellyswitch25-BCDDC27768EF.local/ota?url=http://rojer.me/files/shelly/2.6.0/shelly-homekit-Shelly25.zip
{"status":"updating","has_update":false,"new_version":"20201128-102046/v1.9.2@e83f7025","old_version":"20201128-102046/v1.9.2@e83f7025"}
waiting for shellyswitch25-BCDDC27768EF to reboot...

Error: HTTP Error 404: Not Found
still waiting for shellyswitch25-BCDDC27768EF to reboot...

Error: HTTP Error 404: Not Found
Traceback (most recent call last):
  File "./flash_shelly.py", line 574, in <module>
    device_scan(args.hosts, action, args.do_all, args.dry_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, stock_release_info, homekit_release_info)
  File "./flash_shelly.py", line 503, in device_scan
    parse_info(device, action, dry_run, silent_run, mode, exclude, version, variant, stock_release_info, homekit_release_info)
  File "./flash_shelly.py", line 466, in parse_info
    write_flash(host, lfw, dlurl, cfw_type, mode, requires_upgrade)
  File "./flash_shelly.py", line 301, in write_flash
    if info['stock_model'] == 'SHRGBW2':
TypeError: 'NoneType' object is not subscriptable
zaventem@Ress-MacBook-Pro tools % ping rojer.me
PING rojer.me (35.205.201.239): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2

Gio-88, 2020-11-28:
Hi, I have this problem.
I am flashing a shelly 2.5 (firmware 1.9.0) for installing homekit "http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip" ( 3 weeks ago i did the same operation with another shelly 2.5 and it's ok); but now there is a error.
I have tried to do the downgrade with the firmware 1.8.3 ( August 2020) but i have the same problem again. 
Can you help me?

thanks!
Giorgio

knabalex, 2020-11-28:
Same problem

Perilax-De, 2020-11-28:
I have the same problem. OTA flashing of a shelly 2.5 is not possible in the moment.

rojer, 2020-11-28:
interesting, 2.5.0 can be flashed but 2.6.0 cannot.
as an immediate workaroud please use `http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.5.0/shelly-homekit-Shelly25.zip`
i am investigating why direct update to 2.6 does not work

istefanpop, 2020-11-28:
Thanks Rojer, 2.5.0 worked flawlessly...

Updating from 2.5.0 to 2.6.0 also worked using the python script!

Kind regards,

Stefan

rojer, 2020-11-28:
found the bug, pushed 2.6.1 with the fix. should be fine now.

closed 2020-11-28 by rojer

acap86, 2020-11-28:
Great thank you very much!

Alfniator, 2020-11-29:
Thanks for the speedy response, however it doesn't work for me:
Last wednesday I Flashed my Shelly2.5 with succes, but now it won't load FW.

I have tried several things, all get same message and don't get new FW:
1.) http://192.168.1.167/ota?url=http://rojer.me/files/shelly/2.5.0/shelly-homekit-Shelly25.zip
2.) http://192.168.1.167/ota?url=http://rojer.me/files/shelly/2.6.0/shelly-homekit-Shelly25.zip
3.) http://192.168.1.167/ota?url=http://rojer.me/files/shelly/2.6.1/shelly-homekit-Shelly25.zip

Please advise, thanks in advance.

andyblac, 2020-11-29:
> Thanks for the speedy response, however it doesn't work for me:
> Last wednesday I Flashed my Shelly2.5 with succes, but now it won't load FW.
>
> I have tried several things, all get same message and [...]

what version are you currently on ?

Alfniator, 2020-11-29:
Current version: 20201128-102046/v1.9.2@e83f7025
You have latest version on your device!

andyblac, 2020-11-29:
@Alfniator is it on DHCP or static IP? also make sure it has live internet connection.

Alfniator, 2020-11-29:
Thanks, is was static. But I noticed I made a typo in de default gateway / DNS. So my fault, there was no internet connection. Didn't notice that.

Alfniator, 2020-11-29:
I have another question, but I will make a new issue for that if I can't find anything in the issue list.

andyblac, 2020-11-29:
Np, glad you got it sorted