mongoose-os-apps/shelly-homekit #299: RPC call for garage opener not implemented.

svn2208, 2020-11-28:
Hi,
I am using a Shelly-1 as GarageOpener via Homekit and it works great. The downside, if I am outside I want to use a VPN and RPC call to open the garage. 
Sadly I discovery, that the code is not allowing that.

Only switches, outlets or locks are allowed. Is there special reason ?

    switch (c->type()) {
      case Component::Type::kSwitch:
      case Component::Type::kOutlet:
      case Component::Type::kLock: {
        ShellySwitch *sw = static_cast<ShellySwitch *>(c);
        sw->SetState(bool(state), "web");
        mg_rpc_send_responsef(ri, nullptr);
        return;
      }

rojer, 2020-11-28:
not really, just didn't have the need.

svn2208, 2020-11-28:
Is it possible to add this in a future version?

rojer, 2020-11-28:
yes, makes sense

rojer, 2020-12-13:
in fact, you can send a "toggle" command via RPC like so:

```
curl http://shelly1-test.local/rpc/Shelly.SetConfig -d '{"id": 1, "type": 5, "config": {"toggle": true}}
```

closed 2020-12-13 by rojer