mongoose-os-apps/shelly-homekit #301: shelly i3 overheating

b2a3e8, 2020-11-28:
first of all: thanks to all contributors for their work on this firmware!

i have three shelly i3 installed at my home and one of them is overheating about 2 times a day. most of the time the shelly is +- 87 degrees celsius but sometimes it's 100 degrees celsius. i can't imagine that this is a proper behaviour. unfortunately when reverting to stock firmware there is no temperature shown in the web-gui or in the rest-api (i only see temperature_status in rest which currently shows "high"). the other two shelly i3 are also +- 85 degrees most of the time, but they're not overheating (as much).
i'm not feeling well with this installation. i don't know if this is an issue with this firmware or maybe something else. or is it working as designed?

I wrote a little python script to monitor the temperature over time (with whelly-homekit-firmware installed again). the script makes a call to /rpc/Shelly.GetInfo and writes sys_temp as well as overheat_on to a file. please find the output below:

```
date-time;overheat_on;sys_temp
11/28/2020-21:59:03;False;87
11/28/2020-21:59:33;False;87
11/28/2020-22:00:03;False;87
11/28/2020-22:00:34;False;87
11/28/2020-22:01:04;False;87
11/28/2020-22:01:34;False;87
11/28/2020-22:02:04;False;87
11/28/2020-22:02:34;False;87
11/28/2020-22:03:05;False;87
11/28/2020-22:03:35;False;87
11/28/2020-22:04:05;False;87
11/28/2020-22:04:35;False;89
11/28/2020-22:05:05;False;91
11/28/2020-22:05:36;False;92
11/28/2020-22:06:06;False;93
11/28/2020-22:06:36;False;93
11/28/2020-22:07:06;False;95
11/28/2020-22:07:36;False;95
11/28/2020-22:08:07;False;96
11/28/2020-22:08:37;False;96
11/28/2020-22:09:07;False;97
11/28/2020-22:09:37;False;97
11/28/2020-22:10:07;False;97
11/28/2020-22:10:38;False;97
11/28/2020-22:11:08;False;97
11/28/2020-22:11:38;False;98
11/28/2020-22:12:08;False;98
11/28/2020-22:12:38;False;99
11/28/2020-22:13:09;True;98
11/28/2020-22:13:39;True;99
11/28/2020-22:14:09;True;99
11/28/2020-22:14:39;True;98
11/28/2020-22:15:09;True;99
11/28/2020-22:15:39;True;99
11/28/2020-22:16:10;True;97
11/28/2020-22:16:40;True;96
11/28/2020-22:17:10;True;95
11/28/2020-22:17:40;True;94
11/28/2020-22:18:10;True;93
11/28/2020-22:18:41;True;92
11/28/2020-22:19:11;True;92
11/28/2020-22:19:41;True;92
11/28/2020-22:20:11;True;90
11/28/2020-22:20:41;False;91
11/28/2020-22:21:11;False;91
11/28/2020-22:21:42;False;90
11/28/2020-22:22:12;False;89
11/28/2020-22:22:42;False;89
11/28/2020-22:23:12;False;89
11/28/2020-22:23:42;False;89
11/28/2020-22:24:13;False;89
```

and here the script:
```python
import requests
import json
import datetime
import time

while True:
  r = json.loads(requests.get('http://192.168.1.11/rpc/Shelly.GetInfo').text)
  f = open(r["name"] + ".log", "a")
  f.write(datetime.datetime.now().strftime("%m/%d/%Y-%H:%M:%S") + ";" + str(r["overheat_on"]) + ";" + str(r["sys_temp"]) + "\n")
  f.close()
  time.sleep(30)
```

b2a3e8, 2020-11-28:
update: it looks like the shellies are getting hotter after i pressed the button and the state is 'on'. they're cooling down if I press again an the state is 'off'. is this usual?

rojer, 2020-11-28:
i have noticed that i3 runs significantly hotter than others. i think it's a hardware issue.

rojer, 2020-11-28:
there may be slight change to how we measure temperature - i learned that i used not quite the exact right type of NTC curve for conversion, that may adjust numbers a bit (not sure whether down or up yet), but shelly i3 is definitely hotter than others during operation, no doubt about it.

b2a3e8, 2020-11-28:
@rojer thanks for your quick response. i just monitored the same behaviour on stock firmware (temp. going up from ‘normal’ to ‘very high’ when switch is on and temp. going back down when switch is off).
i think so too that this is an hardware-issue..

rojer, 2020-11-28:
right. go ahead and report it in the official shelly user group then.

thanks!

closed 2020-11-28 by rojer

b2a3e8, 2020-11-28:
i'll do that. thanks for your help!!!

kstrid, 2020-11-28:
Question. Is it getting hotter using momentary button or are you using normal switch button?

b2a3e8, 2020-11-28:
i'm using normal switch buttons

kstrid, 2020-11-28:
Think problem is solved with momentary button, although it should work with normal switch without overheating. Ordered I3 yesterday, eager to try out.

b2a3e8, 2020-11-28:
yes i think so too, but that doesnt help me (-;

b2a3e8, 2020-11-28:
just created a thread in the shelly-support-forum:
https://www.shelly-support.eu/forum/index.php?thread/6002-shelly-i3-system-temperature-issues/

kstrid, 2020-11-28:
I can see you posted on Shelly support under the 1L but it is I3 you're having a problem with, right?

b2a3e8, 2020-11-29:
yes its the i3 im having troubles with. i posted under shelly1 because no shelly i3 category/forum is available

rojer, 2020-12-03:
FYI, in 21fde4820cbf33ab745786aa6c66d3d5e581256d i updated the curve to match the exact part used by Shellies (thanks @kzyapkov for providing the information), values are more accurate now and generally lower in absolute terms.