mongoose-os-apps/shelly-homekit #304: Roller Shutter mode shows wrong excess Input setting (Input 2)

Alfniator, 2020-11-29:
I have a problem with the Roller Shutter mode. i could not find anything in the issue list, I will describe this in detail so other persons can find it on my key words.

Shelly 2.5 mode Roller Shutter
Problem, there is an extra Input called -->    Input 2 (Input 2)
See picture 1 (wrong)                                           -->    General > Roller shutter > Input 2 (Input 2) > HomeKit > WiFi
The other picture shows the correct setup, only  -->    General > Roller shutter > HomeKit > WiFi

How can I get rid of the Input 2 setting? This is also appearing in HomeKit as an extra input, while there is no extra input.

Please advise, thank you.

Wrong situation
[image]

Correct situation
[image]

jobl1978, 2020-11-29:
Change your input mode to one of the separate modes

Alfniator, 2020-11-30:
Thanks for you reply.
Then there is something else going on, can you tell me what your thoughts are on this?

Luifel 3 --> RollerShutter Nr3 --> Input mode Single --> hardware-Single-switch is connected to SW1 input.
This device does exactly what I want it to do (except it shows Input 2 while I want to disable it to export to HomeKit, but this is minor).
The correct behavior: Press (1) hardware-Single-switch, Roller Shutter opens, Press (2) again, Roller Shutter stops, Press (3) again, Roller Shutter closes.

Luifel 2 --> RollerShutter Nr2 --> Input mode Single --> hardware-Single-switch (used same as above) is connected to SW1 input.
This device does not what I want. Press (1) hardware-Single-switch, Roller Shutter opens, Press (2) again,  Roller Shutter stops, Press (3) again, Roller Shutter opens.
When I press (4) again, the Roller Shutter stops, Press (5) again, Roller Shutter opens. Apparantly the Roller Shutter hangs in the Open-Stop-Open-Stop-Open sequence...

Any ideas on this?
The Shelly's 2.5 of Luifel 3 and 2 are 100% exactly installed and configured. Only difference is that I have configured Luifel3 (correct one) last week Tuesday (24-nov) and the Luifel2 last Saturday (28-nov).

**Roller Shutter 3:**
[image]

**Roller Shutter 2:**
[image]

lucaspinelli85, 2020-12-03:
Hi, i try it in configuration sw1 to single momentary for roller shutter, sw2 to input 2 bridge. But if they press contemporary don't work anything... is it right?

Alfniator, 2020-12-03:
Very strange, but you are right.
I just swapped the setting to momentary and back to single, now it works.
The wall-switch properly does Open>Stop>Close.
Also, now the input type 2 is visible in WebAPI.
Thanks! How did you figure this out?

What is good practice now? This is an issue that needs solving right?

lucaspinelli85, 2020-12-03:
Problem is in the 2 input there are 2 physical buttons, one for the roller shutter (open, stop, close...), the other for another thing your choice in the bridge. If I press simultaneously both buttons nothing is happening...

@rojer hear what I thought, I want to connect only one button in both input (1 and 2), when I press the button close the circuit in both SW for this reason: input 1 need for roller shutter in single momentary mode (open, stop, close, stop...), input 2 need for bridge in the app, being that i don't really know if the shutter will go down or up (**I don't see it in the dark!!!!!**), so I thought to set in bridge (input 2) single press no action (it it would conflict with input 1), double press for open roller shutter, hold press for close roller shutter. It would be great so I know what direction to take to the shutter when I don't see it!!!!!!!! (If I use only input 2 I lose the stop movement)
Result is this: 
- single press cyclic with input1 (input2 bridge disabled) open, stop, close, stop... (bridge can't do this)
- double press thanks to bridge open roller shutter
- hold press thanks to bridge close roller shutter
what do you think about it? it is feasible?

rojer, 2020-12-03:
@lucaspinelli85 how many physical switches do you have? what you are describing is possible with two separate switches. or do you want one switch to do all this?

lucaspinelli85, 2020-12-03:
Yes it is possible with two physical buttons, but I don't have space for two, only one module and I can't put a single module with two buttons up and down, I can only put a toggle, a momentary or an edge switch. For this reason I think to put a momentary switch with multiple action

rojer, 2020-12-03:
@lucaspinelli85 ok, i understand. basically, you want additional functions in single input mode for the double and long press: double to open, long to close. this is neat, i kinda like it, i think we can do it, it won't conflict with the cycle mode on single press, so why not.
can you please fiel a separate issue for it though?

meanwhile, i';m going to close this one as it seems the problem has been solved.

closed 2020-12-03 by rojer

lucaspinelli85, 2020-12-03:
> can you please fiel a separate issue for it though?

Ok!