mongoose-os-apps/shelly-homekit #306: HomeKit Secured Router

RedAnon, 2020-11-30:
I've just set up my first shelly (Shelly 1 with FW 2.6.1) and connected it to my Linksys Velops.
I have no idea if the current behavior is correct or not and I haven't found any further info so here is its behavior with SR:
It shows as "Allterco Switch" and with "Restrict to Home" you can't connect to the web UI but remote access works (Apple TV used as a hub). "Automatic" and "No Restriction" are essentially the same and work without any limitations.
IMHO "Automatic" is basically a "No Restriction" mode but with some rules applied (and since this is an unofficial FW I bet there is no way to submit a rules list to Apple, hence no rules applied with "Automatic").
TBH for me it works as intended and I'll be running mine on "Restrict to Home" and switch them to "No Restriction" only for FW update. But perhaps the web UI should be accessible from local network? Also maybe there is a way to get a rules list to Apple somehow?

hughfr4nc15, 2021-01-04:
Hello,

I also Have Velops and was wondering if I should include them in Homekit. Now that one month has passed what's your experience with it? Any issues? You're still unable to access the Web UI? I'm "afraid" that this will create some kind of connectivity issues...

RedAnon, 2021-01-04:
Hi,
0 issues so far with the "Homekit secure router" feature.
The only issue I've experienced with Shelly-Homekit (I have several 1s, 2.5s, and a few plugs) so far is that if the power goes out not all of them come back online and I have to manually restart them. I believe it has something to do with Shellys connecting only the to main Velop AP (I have 4 of them around the house). Which I believe should be solved by the 2.7.0 firmware which I haven't updated to yet.
But yeah, the "Restrict to Home" option (which I use almost exclusively) only allows connections trough the Homekit app (and from the outside through my Apple TV Homekit hub). As far as I can tell the "Automatic" and "No Restriction" modes are essentially the same and always will be as there is no way to add exceptions manually and I don't believe Apple will allow rules for 3rd party FOSS firmwares.
So TL:DR only downside is the need to turn it off when using the WebUI or updating.

Also beware that I'm paranoid and use all the security settings I can and I accept the occasional downsides. Also have completely different, hardware separated networks for my Apple devices (Velop) and other, truly secured devices (OpenWRT). And I would recommend that to everyone.

hughfr4nc15, 2021-01-04:
Thank you for your answer. Will test mine out.

andyblac, 2021-03-03:
can this be closed now ?

hughfr4nc15, 2021-03-03:
Ok by me.

closed 2021-04-11 by github-actions[bot]