mongoose-os-apps/shelly-homekit #308: Wipe all configuration

rudyemm, 2020-11-30:
Is there a way to fully wipe a Shelly 2.5 without physical access to it? I would like to perform the reset described in Recovery > From crash looping or otherwise misbehaving firmware:

In this mode device wipes all its configuration, HAP and WiFi and reverts back to AP mode. Bare-bones web interface will be available at http://192.168.33.1/ when connected to the device's AP, it allows uploading new firmware (can be HomeKit or stock).

rojer, 2020-11-30:
there isn't a way right now, but i think it makes sense to add it.

andyblac, 2020-11-30:
yeah, rojer and i discussed this the other day.

andyblac, 2020-11-30:
the question remained how to do it ?

rojer, 2020-11-30:
it's easy actually, RPC + button in the web ui

rojer, 2020-11-30:
@andyblac what we discussed was how to trigger it physically on shelly1, since it has no physical button.

rudyemm, 2020-11-30:
That’s great!

Will this type of reset go back to the stock firmware? Or stay on HomeKit?

rojer, 2020-11-30:
no, that would need to be done separately.

rudyemm, 2020-12-01:
Do you know, is there a similar function in the stock FW to wipe all config? Even when doing a factory reset in the stock FW, it seems the config remains.

rudyemm, 2021-01-04:
Is this feature something that can be incorporated into the coming builds? 🙏

rojer, 2021-01-06:
at some point - sure

rojer, 2021-01-12:
fixed - added a "Reset" button to the system section, it wipes all the configuration and reboots the device into AP mode.

closed 2021-01-12 by rojer