mongoose-os-apps/shelly-homekit #309: Add Input call to the REST API.

andyblac, 2020-11-30:
could we add `Input` call to `http://shelly1-test.local/rpc/Shelly.SetSwitch?id=1&state=true`, so when actioned it send a fake input call (single, double, long press), which in turn would call the action linked to it in HomeKit

I have shelly button1, but the shelly-homebridge plugin does not work with it very well, so my idea was use the stock firmware and use the API calls for each press action, I have a shelly1 in input mode that performs homekit actions, I could then link these action and have portable button for those actions.

Example:
used as a remote button for opening soma shades (these are battery operated shades) with no physical controls, I have my shades to input 2 on a shelly2.5.

rojer, 2020-12-05:
try 2.7.0-alpha5

andyblac, 2020-12-05:
i get this
```
andrews-mbp:tools andrewblackburn$ curl http://Landing-Alarm-SW.local/rpc/Shelly.InjectInputEvent?id=1&event=2
[1] 5524
andrews-mbp:tools andrewblackburn$ {"code":400,"message":"id and event are required"}

[1]+  Done                    curl http://Landing-Alarm-SW.local/rpc/Shelly.InjectInputEvent?id=1
```

andyblac, 2020-12-05:
ok it needs to be quoted. but it still does not work for shelly url actions

[image]

rojer, 2020-12-05:
shelly can't resolve .local names, you'll have to use ip address

andyblac, 2020-12-05:
yup all working.

closed 2020-12-05 by rojer