mongoose-os-apps/shelly-homekit #310: Shelly stability on weak WiFi

rudyemm, 2020-11-30:
I thought I would pull this out from #252 as a separate topic. In my experience over a couple months, a weak WiFi signal wreaks havoc in the stability of the Shelly – resulting in the web UI being completely inaccessible and the Shelly FW crashing and rebooting. This behavior does not seem to occur on the stock FW.

Is it possible to thoroughly test the Shelly 2.5 on weak (RSSI: -80) WiFi to determine the issue and help improve the FW?

jllavina, 2020-12-04:
Hi. I have a strange behavior that could be related with this post. I have installed 3 shelly 2.5 to control the roller shutters at home. 2 of them are working without problems, but the third only works for a couple of days and then it loses the connection to the Wi-Fi. The phisical switches are still working without problems, but the only solution to recover the device is to cut off the power to force a restart. Then, it reconnects to the wi-fi immediately.

rudyemm, 2020-12-04:
This might also be related to issue #252 – there’s a few of us reporting issues with the Shelly 2.5. My goal here was to test/analyze how a Shelly 2.5 behaves with a poor WiFi signal. Does it become unresponsive? Does it enter a reboot loop?

rojer, 2020-12-19:
@rudyemm is it any better for you on 2.7?

rudyemm, 2020-12-21:
Unfortunately not. Maybe this is anecdotal, but I find the Shellys crashing/rebooting much more often on beta 2 than beta 1 or 2.6.1.

andyblac, 2020-12-21:
> Unfortunately not. Maybe this is anecdotal, but I find the Shellys crashing/rebooting much more often on beta 2 than beta 1 or 2.6.1.

can you supply logs?

rojer, 2020-12-22:
i think i may have found the reason for this, please install 2.7.0-beta3 and let me know if it helps - #330

andyblac, 2021-01-04:
@rudyemm how is 2.7.0 performing ?, any better ?

rudyemm, 2021-01-04:
Unfortunately, not any better. One Shelly has been up for 9 days, another has been up for a few hours, and the rest are in between. I had set up this topic in relation to my Shelly that had the weakest WiFi and was the most unstable, but even my Shellys that have a good connection are still crashing/rebooting, so I think there's still more unknown issues related to #252

rojer, 2021-01-06:
please test 2.7.1, i fixed a whole bunch of bugs that could have affected stability.

rojer, 2021-01-17:
@rudyemm in version 2.7.2 i added a workaround - device will reboot if there is no wifi connection for 5 minutes.

pnuding, 2021-02-15:
Just to chime in that this remains a headache. After a successful pilot I've switched 25 shellies here at home to this project (2.7.3) and while all of them were rock-solid with uptimes in months on the stock FW, they're now a mixed bag. 18 of them have been running for a week without a hitch. 6 have had a few hiccups and 1 (which gets poor signal strength) is rebooting a dozen times every day.
While I do have a mesh network (3 wired AVM Fritz APs, configured to fixed 2.4GHz channels), the moments of instability show no correlation with mesh changes (no AP steering, no other devices connecting/disconnecting around the time of the the problems, other shellies which are literally centimeters away have no problems) and sometimes the affected shelly is just 2 meters away from an AP. 
I also noticed that it's often the same Shelly rebooting several times within an hour or so, then being stable again for several days.

pnuding, 2021-03-30:
This morning one of the finicky Shellies here threw a major fit, which i think could be interesting. 
As mentioned earlier they are sometimes unstable and sometimes not. This one had been running fine for a few days despite reporting a fairly poor signal of -86dBm (not sure why. the one 3 centimeters to the left reports -70) 
All Shellies are on 2.8.2, Wifi settings and router/AP firmwares haven't changed recently.
The Shelly this post is about is a 2.5, has one Osram LED bulb and a heated mirror (total ~270W) on Output 1, and 3 Osram LED bulbs on Output 2 (~12W total).

Then this morning while I was in the shower and had the lights on this one went nuts. Reboots galore, in random intervals, never stable for more than a minute, some of reboots after barely 10 seconds of uptime. It was also running hot because of the reboots (as high as 79ºC, while normally it's around 60)
-> so this is **not** down to the device not getting wifi and timing out!
I saw it responded fine to the switch inputs. If I turned off either of the outputs it would suddenly be stable, only would go nuts if both are on together.
Now 6 hours later it reports the same RSSI strength, and seems stable no matter what i turn on.

pnuding, 2021-04-14:
Confirmed with closeby repeater that improving the wifi strength is making this unstable shelly stable, even when both outputs are on. 
This makes me think: Is this firmware perhaps doing something with heavy calculations at time when the esp8266 is also spiking with wifi power usage? If both output relays are on, perhaps the total consumption is pushing the Shelly 2.5's power supply over the edge and that's what's causing the mad repeated reboots?

DarkStarDS9, 2021-08-06:
Chiming in on this - Shelly 2.5 with Firmware 2.9.5 in Garage Door Opener mode, supplied by the opener (24V). RSSI = -79. If you manage to open the Web-UI, opening/closing works fine, repeatedly. Seems to work fine with HomeKit too. Leave it alone for a few minutes and reload the Web-UI: timeout, apparently rebooting. Takes a few retries to connect after that - usually getting the Shelly-Homekit header and "loading...", but nothing else. Try to reload a few times, and it will be back and seemingly stable as long as you are connected to the UI.

Log-excerpt:

883586097 shelly_main.cpp:543     Up 883.57, HAP 0/2/12 ns 2, RAM: 24476/39048; st 58; 5.1: cur:closed tgt:closed cl:1 op:-1
891107464 mgos_http_server.c:180  0x3fff5fac HTTP connection from 192.168.1.104:61033

it apparently reboots right after connecting.

DarkStarDS9, 2021-08-07:
After leaving it alone for the night and directly connecting to the log-link this morning:
11812304044 shelly_debug.cpp:227    No log file, sending new entries
11812311428 mgos_http_server.c:180  0x3fff5514 HTTP connection from 192.168.1.112:56144
11816589054 shelly_main.cpp:543     Up 11816.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11824589560 shelly_main.cpp:543     Up 11824.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11832589163 shelly_main.cpp:543     Up 11832.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11840589377 shelly_main.cpp:543     Up 11840.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11848589190 shelly_main.cpp:543     Up 11848.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11856589187 shelly_main.cpp:543     Up 11856.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
11864589237 shelly_main.cpp:543     Up 11864.58, HAP 0/2/12 ns 2, RAM: 24192/39048; st 57; 5.1: cur:closed tgt:closed cl:1 op:-1
<rebootet again>

I pulled a coredump yesterday.

rojer, 2021-08-08:
inteersting, can you share privately with me? rojer@rojer.me

rojer, 2021-08-09:
the core dump you sent was corrupted, let's wait until a full one gets written.

rojer, 2021-08-28:
wow, you went deep :)
yeah, i'm also surprised why stock fw wasn't affected. it runs exactly the same wireless stack and no special settings, as far as i know. a mystery.

anyway, i'm going to close this issue, thanks for providing an update!

closed 2021-08-28 by rojer

DarkStarDS9, 2021-08-29:
Hm guess I have to check mine too. Still, while troubleshooting this issue, I noticed other (d)effects:

1. I could not pull a single non-corrupted coredump yet. Since the JSON itself seems to be corrupted, and I assume that this is generated while downloading and not while writing the dump, this looks like a bug, right?
2. While trying to see if a better signal would improve things, I noticed that the Shelly's AP was active, even though it was connected to my network at that time. Guess that should not happen either, or is there any condition where this is expected?