mongoose-os-apps/shelly-homekit #311: Shelly 1 stop reaponding

moonloopsun, 2020-12-01:
Hi guys. Great project!!!!
I configured Shelly 2.5 for my shutters and it works perfectly with HomeKit and with my hue switch configured through homekit. 

I experience a weird behavior from my Shelly 1. 
it’s connected to homekit and configured to turn on via a hie switch. 
The switch works great for a few minutes and than stops working, while the shelly still works through the homekit app. 
When i restart the shelly or the homekit hub, the switch working again. 

Any idea?
Thanks a lot fir your time.

rojer, 2020-12-01:
please enable debug log in the web ui, save it and post here.

moonloopsun, 2020-12-01:
Thanks a lot. I enabled debug log, but struggle to find how to access it. 
Would you kindly explain?

moonloopsun, 2020-12-01:
Great. Thanks a lot for the answer. 
How long would you like the log to run?

moonloopsun, 2020-12-01:
Here is the first log. 

3867723467 shelly_main.cpp:554     Up 3867.64, HAP 0/2/9 ns 2, RAM: 26336/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3875655503 shelly_main.cpp:554     Up 3875.64, HAP 0/2/9 ns 2, RAM: 26340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3883655896 shelly_main.cpp:554     Up 3883.64, HAP 0/2/9 ns 2, RAM: 26340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3891655561 shelly_main.cpp:554     Up 3891.64, HAP 0/2/9 ns 2, RAM: 26340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3899655785 shelly_main.cpp:554     Up 3899.64, HAP 0/2/9 ns 2, RAM: 26340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3905848083 mgos_http_server.c:180  0x3fff3cc4 HTTP connection from 192.168.2.102:56225
3906081998 shelly_debug.cpp:203    Tailing
3907656141 shelly_main.cpp:554     Up 3907.64, HAP 0/2/9 ns 2, RAM: 25832/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3915655760 shelly_main.cpp:554     Up 3915.64, HAP 0/2/9 ns 2, RAM: 26336/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3922705313 mgos_http_server.c:180  0x3fff56ac HTTP connection from 192.168.2.102:56239
3922731139 mgos_http_server.c:180  0x3fff527c HTTP connection from 192.168.2.102:56240
3922746924 mgos_http_server.c:180  0x3fff5b34 HTTP connection from 192.168.2.102:56241
3922929033 mgos_http_server.c:187  0x3fff56ac GET /
3922972686 mgos_http_server.c:187  0x3fff56ac GET /logo.png
3922994614 mgos_http_server.c:187  0x3fff527c GET /axios.min.js.gz
3923021761 mgos_http_server.c:180  0x3fff5bbc HTTP connection from 192.168.2.102:56242
3923034236 mgos_http_server.c:187  0x3fff5bbc GET /logo.png
3923196257 mgos_http_server.c:187  0x3fff5b34 GET /axios.min.js.gz
3923356227 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.2.102:56239
3923377075 mgos_mongoose.c:66      New heap free LWM: 19968
3923656146 shelly_main.cpp:554     Up 3923.65, HAP 0/2/9 ns 2, RAM: 24352/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3931655712 shelly_main.cpp:554     Up 3931.64, HAP 0/2/9 ns 2, RAM: 24352/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3937148255 shelly_debug.cpp:203    Tailing
3939655013 shelly_main.cpp:554     Up 3939.64, HAP 0/2/9 ns 2, RAM: 24340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3947655381 shelly_main.cpp:554     Up 3947.64, HAP 0/2/9 ns 2, RAM: 24340/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3955655081 shelly_main.cpp:554     Up 3955.64, HAP 0/2/9 ns 2, RAM: 25820/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3963655548 shelly_main.cpp:554     Up 3963.64, HAP 0/2/9 ns 2, RAM: 25856/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3971655237 shelly_main.cpp:554     Up 3971.64, HAP 0/2/9 ns 2, RAM: 25856/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3979655056 shelly_main.cpp:554     Up 3979.64, HAP 0/2/9 ns 2, RAM: 25856/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
3987655426 shelly_main.cpp:554     Up 3987.64, HAP 0/2/9 ns 2, RAM: 25856/41288; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0

moonloopsun, 2020-12-01:
It’s super weird. 
That’s the behaviour right now:

The Shelly is configured to be controlled with a hue button in home kit. The home kit hub is my apple tv. 
The shelly works perfectly in the homekit app. 
BUT
works only for a few minutes with the hue button. 
Than it stops responding to the hue button even though i see in the hue app that the button was pressed. The shelly still works through the app. 

In this point i need to restart the homekit hub or the router and the shelly starts responding again to the button for a few minutes. 

Super strange. No?

andyblac, 2020-12-01:
appletv's are realy bad hubs, do you have a HomePod ?

moonloopsun, 2020-12-01:
No i don’t. I have an ipad but experienced the sane behavior with it as well. Same thing....

andyblac, 2020-12-01:
ipad running same same ios ?

moonloopsun, 2020-12-01:
Yes. Sure. All updated. Restarted. Reseted. Logged out and back into icloud........

andyblac, 2020-12-01:
hmm, have tried reboot hue hub ?

moonloopsun, 2020-12-01:
Sure... no change,....

rojer, 2020-12-13:
please help testing 2.7, it comes with updated SDK and a couple stability issues fixed - https://github.com/mongoose-os-apps/shelly-homekit/issues/330

andyblac, 2021-01-04:
@moonloopsun how is 2.7.0 performing, is it any better ?

rojer, 2021-01-06:
i'm going to close this one, we have other similar issues.

closed 2021-01-06 by rojer