mongoose-os-apps/shelly-homekit #312: Web UI Changes over RPC

Rithord, 2020-12-01:
Hello,

Just wondering if anyone can give some advice on how to get MOS tool to work. 

Trying to achieve this command successfully for re-upload:
mos --port ws://x.x.x.x.local/rpc put fs/index.html.gz

I have been trying to update the html web layout of the shelly homekit page and would like to upload it back to the server to make those changes. I have been trying for hours to get MOS Tool working on Windows/Mac and Linux but I'm not having any luck with the instructions that were given in the development section.

Has anyone else done this on their local network yet?

Thanks in advance!

rojer, 2020-12-01:
`ws://x.x.x.x.local/rpc` - this looks wrong. use either ip address, `ws://x.x.x.x/rpc` or the .local name, in which case it's `ws://YYY.local/rpc` where YYY is the device name configured in the UI

Rithord, 2020-12-02:
Thank you for suggesting that, my trouble is actually getting the MOS tool to respond to a RPC call or command. I installed via the instructions that Mongoose OS provides for Windows installations but I was not successful in getting it to work or to communicate with my Shelly.

Any advice here? 

Thank you again!

rojer, 2020-12-02:
please post the command and output you get

Rithord, 2020-12-02:
[image]

rojer, 2020-12-02:
forget gui, use command line.

andyblac, 2020-12-02:
at top, you have not chosen port / board type.

andyblac, 2020-12-02:
> forget gui, use command line.

i was just going to say that too.

Rithord, 2020-12-02:
Is that just using GIT command line?

Rithord, 2020-12-02:
Sorry guys about my lack of knowledge with this

Rithord, 2020-12-03:
> forget gui, use command line.

Hi @rojer, just wondering if you are able to give some advice on how to go about what you mentioned with the command line?

closed 2020-12-03 by Rithord