mongoose-os-apps/shelly-homekit #314: Set code on first install via cli

timoschilling, 2020-12-02:
It would be nice if the flash script would accept a `-c/--code` option with direct set the HomeKit code after the install of the Shelly HomeKit firmware.

andyblac, 2020-12-02:
Please explain more, you mean allow to set HomeKit pairing code ?

timoschilling, 2020-12-02:
Yes I mean the HomeKit pairing code.

Assume I have an devise with stock firmware, connected to my wifi.

Current workflow:
1. run `flash_shelly.py -y shelly-xxx.local`
2. browse to `http://shelly-xxx.local`
3. enter HomeKit pairing code
4. setup devise in the Home app

Requested workflow:
1. run `flash_shelly.py -y --code 11223344 shelly-xxx.local`
2. setup devise in the Home app

I think doing a `POST http://10.0.0.135/rpc/HAP.Setup` request with `{"code":"112-23-344"}` should do the trick.

andyblac, 2020-12-02:
sorry but nope
```
Configuring HomeKit setup code...
requests.post(url='http://192.168.3.16/rpc/HAP.Setup', data={'code': '111-22-333'}
{"code":400,"message":"either code or salt + verifier required"}
```

andyblac, 2020-12-03:
got it done with rojers help

[flash_shelly.py-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5632971/flash_shelly.py-new-build.zip)

this build has had a major rewrite, please report any issues.

timoschilling, 2020-12-03:
Wow that was a fast response / implementation and it works nicely!

But I found one UX glitch, the user needs to use the format `111-22-333` and not `11122333`. If the use uses the wrong format, he didn't get any negative feedback, it only shows: `Configuring HomeKit setup code...`. I would expect a error message or even better that the flash script reformat the number.

And one 💅 point it would be nice to have a handy `-c` or `-h` shortcut for the long `--hap_setup_code`

andyblac, 2020-12-03:
i'll add a format check and report to user how to format if wrong. i'll also add to help.

rojer, 2020-12-03:
we should allow 11122333 format, like UI does

andyblac, 2020-12-03:
ok i'll get script to correct the formating

andyblac, 2020-12-03:
try this added `-c` too @rojer if you prefer not to have this just say.

[flash_shelly-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5636206/flash_shelly-new-build.zip)

andyblac, 2020-12-03:
> And one 💅 point it would be nice to have a handy `-c` or `-h` shortcut for the long `--hap_setup_code`

-h is already used for help info 😉

timoschilling, 2020-12-03:
Tested your latest update and updating a stock shelly1 to HomeKit with setting a code works as expected!

I think you should rename `hap_setup_code` to `hap-setup-code`, CLI args are mostly dasherized

timoschilling, 2020-12-03:
Thanks for the great work!

closed 2020-12-03 by timoschilling

andyblac, 2020-12-03:
done [flash_shelly-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5636253/flash_shelly-new-build.zip)

timoschilling, 2020-12-03:
The two latest versions have a bug, it looks like you always try to convert the code even if not code is given.
```
Traceback (most recent call last):
  File "/Users/captainhagbard/Downloads/./flash_shelly.py", line 601, in <module>
    args.hap_setup_code = f"{args.hap_setup_code[:3]}-{args.hap_setup_code[3:-3]}-{args.hap_setup_code[5:]}" if '-' not in args.hap_setup_code else args.hap_setup_code
TypeError: argument of type 'bool' is not iterable
```

andyblac, 2020-12-03:
sorry about that try this one
[flash_shelly-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5638424/flash_shelly-new-build.zip)

andyblac, 2020-12-03:
oops soz bug in that one
[flash_shelly-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5638441/flash_shelly-new-build.zip)

andyblac, 2020-12-03:
haha, last try
[flash_shelly-new-build.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5638452/flash_shelly-new-build.zip)

timoschilling, 2020-12-03:
tested:
- `./flash_shelly.py`
- `./flash_shelly.py -la`
- `./flash_shelly.py -c 11223344 shelly1-BCDDXXX.local`

every thing works as expected!

nice!