mongoose-os-apps/shelly-homekit #318: option to disable inputs

timoschilling, 2020-12-03:
This feature request targets the i3, but maybe also useful for all devises with switches in detached mode.

I would like to have an option to disable an input in the web ui, to hide them from HomeKit. The reason for this is that I have may i3 installed in my house but all of them only with 1-2 inputs in use. This results in many unused inputs in the Home app. Disabling the unused inputs would be help to keep the overview clean.

rojer, 2020-12-03:
you are in luck, this is already implemented together with motion sensor mode (#294), i'll push test versions soon.

rojer, 2020-12-03:
please try 2.7.0 test version - https://github.com/mongoose-os-apps/shelly-homekit/issues/294#issuecomment-738405359

timoschilling, 2020-12-04:
Works as expected, nice, tanks!

closed 2020-12-04 by timoschilling