mongoose-os-apps/shelly-homekit #319: Additional functions in single input mode for the double and long press

lucaspinelli85, 2020-12-03:
Would be great if in shelly25 in single input mode (open, stop, close, stop...) to exploit input 2 for bridge for double and long press.
In particular to open shutter with double and close shutter with long (already giving direction without waiting for the cycle).
All with the same buttons connected to input 1 and 2.
Result:
- single press: cyclic with input1 in single momentary mode (input2 bridge must be disabled to not conflict) open, stop, close, stop... (bridge can't do this!)
- double press (input2) thanks to bridge open roller shutter
- hold press (input2) thanks to bridge close roller shutter

rojer, 2020-12-03:
seems reasonable and easy to do.

andyblac, 2020-12-03:
could this be coded so it exposes the same 'double press' and 'long press' as stateless switch in homekit.

example:
SW1: wired to light - set as momentary (on/off)
SW2: not wired to anything - linked to bridge of stateless switch for double press and long press (no single press shown)

rojer, 2020-12-03:
this is specifically for roller shutter control

lucaspinelli85, 2020-12-03:
thanks for the correction! sorry for my perhaps incomprehensible language!

andyblac, 2020-12-03:
> this is specifically for roller shutter control

ok, but could it be done ? so we can have single press acting as a relay control, and still have double / long as a stateless switch in homekit.

lucaspinelli85, 2020-12-04:
> ok, but could it be done ? so we can have single press acting as a relay control, and still have double / long as a stateless switch in homekit.

Yes right, because single press in HomeKit can't cycling roller shutter (open, stop, close, stop...) and whereas in the dark I can't see if roller shutter will go up or down, in this way I'm sure is closing or opening (I can't put 2 switches I haven't space). If I did only with HomeKit I'll can only open all and close all, I'll can't stop where I want because bridge is only a scene

lucaspinelli85, 2020-12-04:
so I would take profit by physical relay cycling and wifi HomeKit bridge together! Amazing

lucaspinelli85, 2020-12-09:
Is there a shortcut for single press in roller shutter detached mode to open, stop, close, stop....?

lucaspinelli85, 2020-12-15:
please integrate it, is essential for my unique button, I have to press it many time to identify correct verse to opening/closing of roller shutter. Unfortunately I can't replace with double button. Would be great to command roller shutter differentiating from pressing the button (like bypassing auto off with long press...).