mongoose-os-apps/shelly-homekit #320: Semi-Detached mode

lucaspinelli85, 2020-12-06:
Premise: generally I don't use "detached mode" and I try to avoid it where possible on principle, because if for any reason the wifi does not work (broken modem, interference etc ..), the physical command no longer works. If I have all the home buttons in detached mode, nothing works anymore...
I don't want this to happen, I want if something doesn't work on the wifi, the physical button has to work.
That said, the "detached mode" is very nice and useful for additional functions (double press, long press) to create additional scenes or control other devices in the house not directly connected.
As said before, this mode doesn't work if wifi has problems.
I would like to take advantage of both modes with a "semi-detached mode", leaving the "switch mode" with a single press (so I have the certainty that if the wifi has problems, the physical button works the same) and "detached mode "for double press and long press for additional functions. (Single press must be disabled in HomeKit as it would conflict)
A similar thing is in the #319  issue.
Is this solution feasible in some way?

andyblac, 2020-12-06:
this sound what is being deceased here https://github.com/mongoose-os-apps/shelly-homekit/issues/147

andyblac, 2020-12-06:
@rojer should i close this ticket ?

rojer, 2020-12-13:
we need to fix connection stability issues, i don't want to add something like this as a workaround for that.

closed 2020-12-13 by rojer

vifi42, 2022-06-02:
I want to second that! I already wrote this in another issue a while ago not realizing, that this was already mentioned multiple times - but since everything has been closed I want to reiterate the enhancement-request.
For me it is not about stability but about usability and speed. Programming a toggle on the single-press in HomeKit is possible but the switching-time is enormous. Also non-tech-savvy guests of the house are still able to use the switch as they are used to. Having the single press being handled by shelly itself is quick and easy, having the opportunity to modify the double and long press in HomeKit is just very very useful!