mongoose-os-apps/shelly-homekit #323: Shelly 1L do not load the firmware

dguinaldo, 2020-12-08:
Hi,  I've tried several times to update the shelly 1L using your firmware, but the device as always the factory firmware after the reboot .

Used URL http://x.x.x.x/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip

Output:

{"status":"updating","has_update":false,"new_version":"20201205-115619/v1.9.2_1L@8f181f76","old_version":"20201205-115619/v1.9.2_1L@8f181f76","beta_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}

Thank you for your help.

Best regards.
David

andyblac, 2020-12-08:
make sure device has valid DNS IP (ie it can reach World Wide Web)

ortoberi, 2020-12-09:
Hi, I have same issue: {"status":"updating","has_update":false,"new_version":"20201205-115619/v1.9.2_1L@8f181f76","old_version":"20201205-115619/v1.9.2_1L@8f181f76","beta_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}. Device has a valid IP an it can reach WWW.

BR,
David

Sandokandrea, 2020-12-10:
Similar issue, 
{"status":"updating","has_update":false,"new_version":"","old_version":"20200320-123430/v1.6.2@514044b4"}
where is my mistake?

dguinaldo, 2020-12-11:
Not a DNS or network problem. factory upgrade process work correctly.

andyblac, 2020-12-11:
still waiting for mine to arrive.

DerDirsch, 2020-12-13:
I am having the same situation on my device

rojer, 2020-12-13:
hm. it works for my device and i see many users also running the firmware (update checks from Shelly1L devices), so it works for at least some others too.
unfortunately, there is not much in terms of diagnostics that we can do here, unless someone can attache a serial adapter or send me a device that refuses to install the HK firmware.

andyblac, 2020-12-13:
still waiting for mine to arrive, i'll test once i have it.

rojer, 2020-12-13:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection#wiring - see here for serial wiring. you only need GND, 3V3, TX and RX pins.

freddair, 2020-12-14:
I have the same issue with my Shelly 1L. Is this addressed in 2.7 beta that we may test?

zhoucky, 2020-12-15:
Same thing. The led is blinking while updating, but after that loads the original interface

rojer, 2020-12-16:
is anyone who experiences this able to connect serial and send me console output or (even better) flash dump?

see https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection#wiring and the next section (up to the flash-read command).

zhoucky, 2020-12-16:
My serial is 5v. I ordered a 3,3 one, but waiting for a delivery. When it arrives, I will send you the dump if somebody didn’t meanwhile

andyblac, 2020-12-16:
still waiting for mine to arrive 😠 , has been stuck in the same place in UK depo for 2 days.

Cojaeck, 2020-12-16:
Sadly I cannot help with serial interface data. But my 1L offer a Beta from the manufacturer which installs. But the install of your firmware also still fails on my devices.

freddair, 2020-12-16:
Interestingly I now found out that the Homekit firmware loads successfully on the second 1L out of the very same box. This time I did not update the stock fw before applying the HomeKit fw.

Cojaeck, 2020-12-16:
I did some more tests. You can downgrade the 1L to all versions in the Shelly archive. But with no version I could install the HomeKit version. Strange....

andyblac, 2020-12-16:
have you tried flashing via my script ?

Cojaeck, 2020-12-16:
No. Just via Rojers Links. At the moment I just have an iPhone to do that :-)

andyblac, 2020-12-16:
> No. Just via Rojers Links. At the moment I just have an iPhone to do that :-)

ok

freddair, 2020-12-16:
@andyblac tried your script before and it did not recognize the shelly 1L in question. I run it again and here is the output (the 1L with fw 2.6.1 is the second one that worked just fine, but the first one is not found).

```
Frederiks-MacBook-Pro:~ frederikmueller$ /Users/frederikmueller/Downloads/shelly-homekit-master\ 4/tools/flash-shelly.py -a
Scanning for Shelly devices...
4 Devices found.

Host: http://Shelly1-Fred-Hauseingang.local
Device ID: shelly1-F010D0
IP: 10.0.0.101
Model: Shelly1
Current: HomeKit 2.6.1
Latest: HomeKit 2.6.1
Does not need flashing...

Host: http://Shelly1-Ralf-Kronleuchter.local
Device ID: shelly1-05BEF3
IP: 10.0.0.99
Model: Shelly1
Current: HomeKit 2.6.1
Latest: HomeKit 2.6.1
Does not need flashing...

Host: http://shelly1l-84CCA8AD2233.local
Device ID: shelly1l-AD2233
IP: 10.0.0.170
Model: Shelly1L
Current: HomeKit 2.6.1
Latest: HomeKit 2.6.1
Does not need flashing...

Traceback (most recent call last):
  File "/Users/frederikmueller/Downloads/shelly-homekit-master 4/tools/flash-shelly.py", line 661, in <module>
    device_scan(args.hosts, action, args.do_all, args.dry_run, args.silent_run, args.mode, args.exclude, args.version, args.variant, args.hap_setup_code)
  File "/Users/frederikmueller/Downloads/shelly-homekit-master 4/tools/flash-shelly.py", line 586, in device_scan
    if (deviceinfo.fw_version == '0.0.0' or is_newer(deviceinfo.flash_fw_version, deviceinfo.fw_version)):
  File "/Users/frederikmueller/Downloads/shelly-homekit-master 4/tools/flash-shelly.py", line 359, in is_newer
    vi1 = parse_version(v1)
  File "/Users/frederikmueller/Downloads/shelly-homekit-master 4/tools/flash-shelly.py", line 355, in parse_version
    major, minor, patch = [int(e) for e in v]
  File "/Users/frederikmueller/Downloads/shelly-homekit-master 4/tools/flash-shelly.py", line 355, in <listcomp>
    major, minor, patch = [int(e) for e in v]
ValueError: invalid literal for int() with base 10: '2_1L'
```

andyblac, 2020-12-16:
@freddair can you run it with flag `-v=1` and copy paste output to a file and post please

andyblac, 2020-12-16:
i suspect is the version number format it does not like `v1.9.2_1L`

freddair, 2020-12-16:
> @freddair can you run it with flag -v=1 and copy paste output to a file and post please

sure, here's the output. now it fount the shelly 1L - what does it say?
[script1L-v=1.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5705333/script1L-v.1.txt)

> i suspect is the version number format it does not like v1.9.2_1L

I also tried v1.9.2 and 1.9.0 - did not work either.

andyblac, 2020-12-16:
yeah it was, try this one 
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5705356/flash-shelly.py.zip)

freddair, 2020-12-16:
[script1L-v.1.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/5705398/script1L-v.1.txt)

andyblac, 2020-12-16:
no crash this time, but still failed to flash can open a console log to ip of device by doing this
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#console-logs

and in seperate shell try script again and send log from console

andyblac, 2020-12-16:
grab mos tool from here https://mongoose-os.com/docs/mongoose-os/quickstart/setup.md

freddair, 2020-12-16:
sorry but I am unable to connect to serial... no suitable connector available

andyblac, 2020-12-16:
> sorry but I am unable to connect to serial... no suitable connector available

use IP method.

freddair, 2020-12-16:
```
Frederiks-MacBook-Pro:~ frederikmueller$ mos --port ws://10.0.0.168/rpc config-set debug.udp_log_addr=10.0.0.168:8910
Getting configuration...
E1216 21:58:49.882361    7725 reconnect_wrapper.go:119] [reconnectWrapperCodec to ws://10.0.0.168/rpc; connect in 2.00s] connection error: read tcp 10.0.0.41:59823->10.0.0.168:80: read: connection reset by peer
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:172: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:201: WebSocket connection failed
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:248: 
E1216 21:59:14.108469    7725 reconnect_wrapper.go:119] [reconnectWrapperCodec to ws://10.0.0.168/rpc; connect in 2.00s] connection error: read tcp 10.0.0.41:59844->10.0.0.168:80: read: connection reset by peer
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:172: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:201: WebSocket connection failed
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:248: 
Error: context deadline exceeded
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/codec/reconnect_wrapper.go:141: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/codec/reconnect_wrapper.go:185: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:401: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn_impl.go:167: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn_impl.go:190: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn.go:117: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/config/config.go:86: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/main.go:198: config-set failed
```

freddair, 2020-12-16:
```
Frederiks-MacBook-Pro:~ frederikmueller$ mos --port ws://10.0.0.168/rpc config-set debug.udp_log_addr=10.0.0.41:8910
Getting configuration...
E1216 22:02:30.263919    7755 reconnect_wrapper.go:119] [reconnectWrapperCodec to ws://10.0.0.168/rpc; connect in 2.00s] connection error: read tcp 10.0.0.41:60022->10.0.0.168:80: read: connection reset by peer
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:172: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:201: WebSocket connection failed
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:248: 
E1216 22:02:54.718038    7755 reconnect_wrapper.go:119] [reconnectWrapperCodec to ws://10.0.0.168/rpc; connect in 2.00s] connection error: read tcp 10.0.0.41:60053->10.0.0.168:80: read: connection reset by peer
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:172: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:201: WebSocket connection failed
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:248: 
Error: context deadline exceeded
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/codec/reconnect_wrapper.go:141: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/codec/reconnect_wrapper.go:185: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/common/mgrpc/mgrpc.go:401: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn_impl.go:167: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn_impl.go:190: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/dev/dev_conn.go:117: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/config/config.go:86: 
/private/tmp/mos-20200820-97577-jfaujm/mos-1ec85951e77649f1151bc06a12901faa784a991a/cli/main.go:198: config-set failed
```

Cojaeck, 2020-12-16:
I also tried your script...but keep getting "Failed to lookup online version information"...
 my fault?

andyblac, 2020-12-16:
@freddair ok guess it does not work with stock fw, I'll try serial once I get my Shelly

freddair, 2020-12-16:
how would you connect serial and power at the same time, or would the serial power be sufficient for operation?

andyblac, 2020-12-16:
yeah you just connect serial power, not AC not the same time.

marcoboerner, 2020-12-17:
> I also tried your script...but keep getting "Failed to lookup online version information�"...
> my fault?

I had the same issue, at first I used the script and got the same `ValueError: invalid literal for int() with base 10: '2_1L'` warning. After upgrading Python and Pip to 3.9.0 I got the warning above.

andyblac, 2020-12-17:
> I had the same issue, at first I used the script and got the same `ValueError: invalid literal for int() with base 10: '2_1L'` warning. After upgrading Python and Pip to 3.9.0 I got the warning above.

here is the latest version of script Pull Request waiting. it fixes `ValueError: invalid literal for int() with base 10: '2_1L'` 
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5712684/flash-shelly.py.zip)

andyblac, 2020-12-17:
> I also tried your script...but keep getting "Failed to lookup online version information�"...
> my fault?

that it SSL certificate error try running

`pip install --upgrade certifi`

marcoboerner, 2020-12-17:
> > I also tried your script...but keep getting "Failed to lookup online version information�"...
> > my fault?
> 
> that is SSL certificate error try running
> 
> `pip install --upgrade certifi`

That was my issue, I used the `Install Certificates.command` script inside the Python 3.9 folder to solve it. Somehow it didn't do that correctly right after the Python upgrade. Would it make sense to have the script print out the urllib errors instead of an error message? Adding this to the code helped me out quite a bit. : )

`except urllib.error.URLError as e:
    logger.warning(e.reason)  `

etc.

I also had to change the line"
`self.host = f'{host}.local' if '.local' not in host and not host[0:3].isdigit() else host`
to just
`self.host = host`
because my network doesn't use the .local ending.

marcoboerner, 2020-12-18:
However, the script is now attempting to flash, waits for the device to reboot for quite some time, then fails to flash.

`Failed to flash xwing-ii to 2.6.1

Current: 1.9.2_1L`

andyblac, 2020-12-18:
> However, the script is now attempting to flash, waits for the device to reboot for quite some time, then fails to flash.
> 
> `Failed to flash xwing-ii to 2.6.1
> 
> Current: 1.9.2_1L`

yeah think it stock fw issue, i need to debug with a device and serial connections, should get my delivery tomorrow hopefully.

andyblac, 2020-12-18:
> That was my issue, I used the `Install Certificates.command` script inside the Python 3.9 folder to solve it. Somehow it didn't do that correctly right after the Python upgrade. Would it make sense [...]

i can add that to verbose logging for when `-v=0` is used.

> 
> etc.
> 
> I also had to change the line"
> `self.host = f'{host}.local' if '.local' not in host and not host[0:3].isdigit() else host`
> to just
> `self.host = host`
> because my network doesn't use the .local ending.

you on a domain ?, i can add a new cmd flag to specify own domain if it would help ? `-d=xxxx`

marcoboerner, 2020-12-18:
I'm not on a domain, my router in standard config just assigns hostnames without the .local ending. Maybe it would be enough to check, when a hostname without .local is entered, if it resolves, and only if not to add the .local part?

andyblac, 2020-12-18:
> I'm not on a domain, my router in standard config just assigns hostnames without the .local ending. Maybe it would be enough to check, when a hostname without .local is entered, if it resolves, and only if not to add the .local part?

ok, see what i can do

andyblac, 2020-12-18:
@marcoboerner can you try this
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5713015/flash-shelly.py.zip)

Cojaeck, 2020-12-18:
@andyblac Thx. I did the certificate update and used your latest script which now runs.
It finds the 1L-Devices (and others) but flashing still fails:

_[1mHost: [0mhttp://shelly1l-84CCA8ADDD06.fritz.box
[1mDevice ID: [0mshelly1l-84CCA8ADDD06
[1mIP: [0m192.168.178.155
[1mModel: [0mShelly1L
[1mCurrent: [0mStock 1.9.2_1L
[1mLatest: [0mHomeKit [1;93m2.6.1[0m
Do you wish to flash shelly1l-84CCA8ADDD06.fritz.box to firmware version 2.6.1 (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8ADDD06.fritz.box to reboot...
still waiting for shelly1l-84CCA8ADDD06.fritz.box to reboot...
[1;91mFailed to flash shelly1l-84CCA8ADDD06.fritz.box to 2.6.1[0m
 
[1mHost: [0mhttp://shelly1l-84CCA8AE3827.fritz.box
[1mDevice ID: [0mshelly1l-84CCA8AE3827
[1mIP: [0m192.168.178.156
[1mModel: [0mShelly1L
[1mCurrent: [0mStock 1.9.2_1L
[1mLatest: [0mHomeKit [1;93m2.6.1[0m
Do you wish to flash shelly1l-84CCA8AE3827.fritz.box to firmware version 2.6.1 (y/n) ? y
Now Flashing...
waiting for shelly1l-84CCA8AE3827.fritz.box to reboot...
[1;91mCould not get info from device: shelly1l-84CCA8AE3827.fritz.box
[0m
[1;91mCould not get info from device: shelly1l-84CCA8AE3827.fritz.box
[0m
[1;91mCould not get info from device: shelly1l-84CCA8AE3827.fritz.box_

andyblac, 2020-12-18:
should have mine today 
```
Your Item was received by xxxxxxxx DO on 18-12-2020 and is now due for delivery today. We normally expect to deliver to your address between 11:53am - 3:53pm.
```

marcoboerner, 2020-12-18:
> @marcoboerner can you try this
> [flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/5713015/flash-shelly.py.zip)

I'm not sure if it's a MacOS thing or the way my router assigns hostnames. However, my Mac it seams, does assign itself the .local domain ending but is reachable with or without it. The Shelly 1L does not, the Shelly 1L gets its hostname from my router though. So socket.getfqdn() will return a full domain with .local. Even though it is not used in my network.

andyblac, 2020-12-18:
> I'm not sure if it's a MacOS thing or the way my router assigns hostnames. However, my Mac it seams, does assign itself the .local domain ending but is reachable with or without it. The Shelly 1L [...]

i'm on Mac, in that case it might be easier to add a flag `--no-local-domian` to bypass adding search and adding it
would that work for you ?

andyblac, 2020-12-18:
@marcoboerner can you do a run with `-v=1` and copy and paste info to a file see what scanner is picking up.

marcoboerner, 2020-12-18:
> @marcoboerner can you do a run with `-v=1` and copy and paste info to a file see what scanner is picking up.
> 
> ie
> `./flash-shelly.py -la -v=1`

Okay, I think I know what's happening. The Shelly 1L with stock firmware doesn't allow its hostname to be changed, meaning it will always have the shelly1l-XXXXXXXXXXX.local address. Through my router I can set a different hostname. Those names don't have the .local added. And in my network I can reach Shelly over the original hostname and the one setup in my router.

I assume the best way is to just use the original hostname for the upgrade and then change the hostname within homekit firmware. However other people might run into the same problem if they use a similar setup / router.

Here is the output:

`sudo python3 ./flash-shelly.py -la -v=1                        
Password:
Finding local domain...
app
OS: Darwin
local_domain: local
manual_hosts: []
action: list
mode: homekit
do_all: True
dry_run: False
silent_run: False
hap_setup_code: False
version: False
exclude: None
variant: False
verbose: 1

stock_release_info:{'isok': True, 'data': {'SHPLG-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-1.zip', 'version': '20201124-091454/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-1.zip', 'beta_ver': '20201202-140129/v1.9.3-rc3@50c6ab57'}, 'SHPLG-S': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-S.zip', 'version': '20201124-092310/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-S.zip', 'beta_ver': '20201202-141017/v1.9.3-rc3@50c6ab57'}, 'SHPLG-U1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG-U1.zip', 'version': '20201124-092346/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG-U1.zip', 'beta_ver': '20201202-141054/v1.9.3-rc3@50c6ab57'}, 'SHPLG2-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHPLG2-1.zip', 'version': '20201124-092420/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHPLG2-1.zip', 'beta_ver': '20201202-141133/v1.9.3-rc3@50c6ab57'}, 'SHSW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-1.zip', 'version': '20201124-091217/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-1.zip', 'beta_ver': '20201216-135937/v1.9.3-rc4@146c9bd7'}, 'SHSW-21': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-21.zip', 'version': '20201128-102008/v1.9.2@e83f7025', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-21.zip', 'beta_ver': '20201202-140000/v1.9.3-rc3@50c6ab57'}, 'SHSW-25': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip', 'version': '20201128-102046/v1.9.2@e83f7025', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-25.zip', 'beta_ver': '20201216-140131/v1.9.3-rc4@146c9bd7'}, 'SHSW-PM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-PM.zip', 'version': '20201128-103131/v1.9.2@e83f7025', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-PM.zip', 'beta_ver': '20201202-141212/v1.9.3-rc3@50c6ab57'}, 'SHSW-L': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-L.zip', 'version': '20201205-115619/v1.9.2_1L@8f181f76', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSW-L.zip', 'beta_ver': '20201202-135922/v1.9.3-rc3@50c6ab57'}, 'SHAIR-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHAIR-1.zip', 'version': '20201124-092534/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHAIR-1.zip', 'beta_ver': '20201202-141250/v1.9.3-rc3@50c6ab57'}, 'SHSW-44': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSW-44.zip', 'version': '20201019-110845/v1.6.6_4Pro@bad181e4'}, 'SHUNI-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHUNI-1.zip', 'version': '20201124-093042/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHUNI-1.zip', 'beta_ver': '20201202-141748/v1.9.3-rc3@50c6ab57'}, 'SHEM': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM.zip', 'version': '20201124-092817/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHEM.zip', 'beta_ver': '20201202-141537/v1.9.3-rc3@50c6ab57'}, 'SHEM-3': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHEM-3.zip', 'version': '20201124-092854/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHEM-3.zip', 'beta_ver': '20201202-142834/v1.9.3-rc3@50c6ab57'}, 'SHSEN-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSEN-1_build.zip', 'version': '20200717-112934/v1.7.7@a02db0b8'}, 'SHGS-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHGS-1.zip', 'version': '20201124-093005/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHGS-1.zip', 'beta_ver': '20201202-141710/v1.9.3-rc3@50c6ab57'}, 'SHSM-01': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHSM-01.zip', 'version': '20201124-091539/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHSM-01.zip', 'beta_ver': '20201202-140215/v1.9.3-rc3@50c6ab57'}, 'SHHT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHHT-1.zip', 'version': '20201124-091711/v1.9.0@57ac4ad8', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHHT-1.zip', 'beta_ver': '20201202-140348/v1.9.3-rc3@50c6ab57'}, 'SHWT-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHWT-1.zip', 'version': '20201128-102432/v1.9.2@e83f7025', 'beta_url': 'http://api.shelly.cloud/firmware/rc/SHWT-1.zip', 'beta_ver': '20201202-140437/v1.9.3-rc3@50c6ab57'}, 'SHDW-1': {'url': 'http://shelly-api-eu.shelly.cloud/firmware/SHDW-1.zip', 'version': '20201124-091845/v [... cut ...]
Mode: Stock To HomeKit

parse_info
device_info: <__main__.StockDevice object at 0x7fd0f3a7ad00>
host: shelly1l-XXXXXXXXXXXX.local
device: shelly1l-XXXXXXXXXXXX
model: Shelly1L
stock_model: SHSW-L
colour_mode: relay
action: list
flash mode: homekit
requires_upgrade: False

Host: http://shelly1l-XXXXXXXXXXXXX.local
Device ID: shelly1l-XXXXXXXXXXXXX
IP: XXX.XXX.XXX.XXX
Model: Shelly1L
Current: Stock 1.9.2_1L
Latest: HomeKit 2.6.1
D_URL: https://rojer.me/files/shelly/2.6.1/shelly-homekit-Shelly1L.zip`

andyblac, 2020-12-18:
> Okay, I think I know what's happening. The Shelly 1L with stock firmware doesn't allow its hostname to be changed, meaning it will always have the shelly1l-XXXXXXXXXXX.local address. Through my [...]

ok, i see, i have my shelly1L, but serial lead does not fit the smaller pin header, i have some new cables coming as soon as they arrive, i will take a core dump, and try to flash it OTA, i do not want to try things untill i have RAW dump of current firmware, I did speak to a Shelly Engineer today about and he said they should be no reason that it should not flash via OTA.

rojer, 2020-12-19:
ok, i reproduced and fixed the bug.
it looks like in 1.9 stock firmware started using hardware timer in NMI mode while our firmware uses it in non-nmi mode.
because this register doesn't get cleared during soft reset, during upgrade this causes an issue because interrupt is not properly configured and never fires.
long story short, https://github.com/cesanta/mongoose-os/commit/41f01a2c76d626249d42bfdc4ec8f068e7251d00 is the fix, i pushed 2.7.0-beta2, please test - https://github.com/mongoose-os-apps/shelly-homekit/issues/323

closed 2020-12-19 by rojer

freddair, 2020-12-19:
Now it works great! Thanks a lot @rojer and @andyblac for your help

DerDirsch, 2020-12-19:
I am even not able to load the BETA firmware to my device.
Same behavior like with the released version.

freddair, 2020-12-19:
Did you use the right URL?

http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.7.0-beta2/shelly-homekit-Shelly1L.zip

Cojaeck, 2020-12-19:
> Now it works great! Thanks a lot @rojer and @andyblac for your help

Same from me: you guys rock! 👍 thanks a lot

DerDirsch, 2020-12-19:
> Did you use the right URL?
> 
> http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.7.0-beta2/shelly-homekit-Shelly1L.zip

Yes. This is was I did.
After adding the link, I receive following message and only stock firmware loads.

{"status":"updating","has_update":false,"new_version":"","old_version":"20201205-115619/v1.9.2_1L@8f181f76"}

Regards
Dirk

DerDirsch, 2020-12-20:
I have changed the Shelly from static IP to DHCP and Voila: it worked!
Great job guys.,

Thanks a lot!