mongoose-os-apps/shelly-homekit #324: Shelly 2.5 Roller Shutter Calibration - special case?

AnMiPe, 2020-12-09:
Hi All!

I have an issue not with the roller shutter calibration itself, but especially with my motor.
My roller motor takes 1-2 seconds until it really works starts moving the shutters. I believe it is some kind of motor protection.
This has the effect, that the shelly instantly stops when pressing UP/DOWN after the calibration.
I think it will solve this (special) issue, if the shelly waits a little more time for the power measurement.

BTW: This seems to be an issue with the stock firmware as well. There I cannot calibrate the shutter successfully. 

Thank you!

AnMiPe, 2020-12-23:
Do you need any more information on this issue? #248 seems to be very similar and I'm not alone with this.

andyblac, 2021-01-27:
@pivale can you offer any assistance ?

pivale, 2021-01-28:
@rojer , is there anyway we can give a min time for calibration higher to accommodate this situation?

Actually I think it's something "bigger" and firmware should still keep power for a couple of seconds even if power consumption is 0... It would solve other problems and power fluctuation... (although not for calibration)

I don't see a drawback with this, since motors need to have auto stop anyway for calibration to work, so no strain on motors should happen

AnMiPe, 2021-01-29:
In my opinion and experience with my roller shutter motor 2-3 seconds will be enough.

andyblac, 2021-03-03:
@rojer can you advise or is https://github.com/mongoose-os-apps/shelly-homekit/issues/404 better solution ?

rojer, 2021-03-06:
i increased it to 5 seconds by default, please install 2.8.0-alpha4 and help test

https://github.com/mongoose-os-apps/shelly-homekit/issues/498

closed 2021-03-13 by github-actions[bot]