mongoose-os-apps/shelly-homekit #325: Wifi Blacklist

up256, 2020-12-09:
Hi guys,
thanks to everyone who worked on this great project. I'm running shelly-homekit on 5 Shellies 2.5 and usually everything works fine. 
In my flat I can see some 30 wifi networks, some of them are not encrypted. Sometimes it happens that one of my Shellies connects to one of the free wifi networks. I can still operate the Shelly over the wall switch but it ist not reachable via homekit. I can't find in my router. The status LED is off, so it must be connected to a wifi, but not mine.

Is it possible to implemet a wifi Blacklist so that the Shelly ignores the wifi networks on that list?

rojer, 2020-12-13:
do the networks have the same name as yours? shelly will only connect to network that matches the name you configured.
in any case, please try 2.7.0 pre-release, it should improve connection stability - https://github.com/mongoose-os-apps/shelly-homekit/issues/330

up256, 2020-12-18:
No, the networks do not have the same name. I actually don’t really know that my shelly was connected to one of the other networks. It’s just a guess because the status LED was off.
But anyway, since a couple of days I’m running 2.7.0 pre-release and it’s working without any issues. So I’m closing this issue. Thank you rojer for your work in this project.

closed 2020-12-18 by up256