mongoose-os-apps/shelly-homekit #326: Script multiple calibration roller shutter

lucaspinelli85, 2020-12-09:
Hi, is there a script that calibrates all shelly roller shutter? Is it necessary to calibrate or it work the same? Thanks

andyblac, 2020-12-09:
No, I'll have a chat with @rojer to see if this is even possible.

So atm you will have to do them one by one.

lucaspinelli85, 2020-12-09:
I must to do it or not necessary? Can i do only in webui or can i one by one from script, url, anything without enter in webui?

andyblac, 2020-12-09:
i am not totally familiar with roller shutter mode, but, from what i understand yes, it must be done, and it can only be done from the webUI.

andyblac, 2020-12-09:
see here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Roller-Shutter-Settings for info.

andyblac, 2021-01-04:
can i close this issue now ?, as it can not be done via a script and mst be done via the webUI.

closed 2021-01-18 by andyblac