mongoose-os-apps/shelly-homekit #328: GDO issue with movement time -- HomeKit state saddening changing during opening to closed

tuffsel44, 2020-12-10:
Hey, I got this issue and took a log file and commented it to explain the issue. Shelly 1 with HomeKit GDO 2.6.1. firmware and a contact sensor at the fully open position. The movement time of the garage is 18 seconds.

The issue is that about 6-8 after opening the garage the HomeKit state is switching from opening to closed suddenly although the garage is still moving and has not reached the contact sensor at the open position. It seems the movement is not correctly recognized.

[image]

[image]

rojer, 2020-12-10:
> contact sensor at the fully open position

this is the problem. we expect a close sensor, i.e. sensor that is activated when the door is fully closed. it can be a normally open or normally closed type, but it's still a "door fully closed" sensor.

tuffsel44, 2020-12-10:
> > contact sensor at the fully open position
> 
> this is the problem. we expect a close sensor, i.e. sensor that is activated when the door is fully closed. it can be a normally open or normally closed type, but it's still a "door fully closed" sensor.

Hey Rojer, is there any way to provide an option in the config to choose if the sensor is located at the opening position? Due to the garage building it is very hard to place my sensor at the close position. That would be perfect. Any chance for this is a future update?

simonfield76, 2020-12-10:
Thank you for all the answer!

Does the problem will be solved setting the shelly as a normal switch?

Inviato da iPhone

rojer, 2020-12-10:
> is there any way to provide an option in the config to choose if the sensor is located at the opening position? 

it wouldn't be difficult but i'm reluctant to do it because it's a less secure configuration. when we have a "closed" sensor, we will only report "door is closed" if we get an activation from the sensor. we don't have a sensor for the open position, so we use a timer and after the door is no longer closed (sensor deactivated), after certain time we report the door as open - we know the door is not closed, so it's probably open. but even if it gets such midway, no big deal, status is "open", so at least we're not telling you it's closed when it isn't.
now, if we replace the sensor with the "open" kind, the situation is opposite: we know for sure when the door is fully open but we have to _guess_ that it's closed when it may not in fact be. the door mat be half open but your app will happily show "closed"

tuffsel44, 2020-12-11:
Understand your point, but why not let the user choose? My usecase is about convenience rather than security. The door will only be opened or closed anyway out the car. The security aspect of what you describe is I guess for most people „nice to have“ but the main reason to use the Shelly in this configuration is that it is convenient. I am unfortunately not able to program it myself nor can I place the sensor at the close position. Maybe you will reconsider and offer an option. That would be fantastic. Thanks.

rojer, 2020-12-13:
sorry, i just don't think it's right to allow this kind of configuration.
you can install an additional contact sensor, they are cheap and it's not difficult.

closed 2020-12-13 by rojer

juangamnik, 2021-01-08:
I really appreciate what you are doing here for us @rojer, but why not letting the user choose? There could be a small warning ⚠️ icon beside the field or something like this to give the user a hint that this is not save... perhaps there is a compromise?

rojer, 2021-01-09:
make code more complicated _and_ setup less secure? this just doesn't sound compelling to me.

juangamnik, 2021-01-09:
It is your project, but from my point of view: there seems to be an audience for that mode and since you made it Open Source it seems to be something that should work for as many people as possible. I will relocate my sensor, so this feature is not for me, don‘t get me wrong.

Thank you for this great software...

rojer, 2021-01-09:
thanks Johannes, appreciate it.

open source actually has nothing to do with how wide the intended audience is. but it does mean that i am openly sharing the results of my labor and it does mean that anyone is free to modify it in any way they see fit. open source is about user enablement - it enables users to become developers too, help themselves and solve their own problems. including making it work with "fully open" sensor. in fact, hacking it to work that way would be pretty easy - basically, flip a bunch of conditions in the code and it's done (proper support, with option etc, would take a bit more work). so, while i am not willing to make it an official feature, anyone is free to do it for themselves.

juangamnik, 2021-01-09:
I know what Open Source is