mongoose-os-apps/shelly-homekit #330: Testing release 2.7

rojer, 2020-12-11:
This issue will gather feedback on the upcoming 2.7 release.

What are we testing:

Ok, here's an alpha version with motion and occupancy sensor modes: http://rojer.me/files/shelly/2.7.0-alpha6/
In addition to stateless switch, detached inputs can be either motion or occupancy sensors. Both level and pulse modes are supported, as requested. @jobl1978 please test and let me know.

 * Motion, occupancy and contact sensor modes for detached inputs (#294 #297)
 * Shelly 1PM, Plug and PlugS power measurement support (#27) 
 * WiFi stability improvements from Espressif SDK update and use of all-channel scan - this should make the device connect to the AP with the strongest signal instead of the first matching the SSID.

Please install binaries from here: http://rojer.me/files/shelly/2.7.0-alpha7/

andyblac, 2020-12-12:
@rojer i'm already on alpha6 ? should it be have been alpha7 or beta1 ?

rojer, 2020-12-12:
i was going to release alpha7 with the new sdk, in fact it was even up for a minute or two, but i found a nast bug when updating my own devices and had to pull it back.
once i fix it and get it more stable, i'll re-publish it.

nliviu, 2020-12-12:
Not related to shelly, but OTA update does not work with the new sdk.

rojer, 2020-12-12:
@nliviu very much related, looking into it.

nliviu, 2020-12-12:
@rojer You are right, other ESP8266 devices do OTA update ok.

rojer, 2020-12-13:
ok, this is ready for testing now. fixed the regression which turned out to be caused by stack overflow.
there's a wider issue there that we have stack overflows happening routinely... i fixed a couple but not all.
this would definitely explain some of the instability. i will look into fixing more of them, but for now alpha7 shoul;d be an improvement already. please test and give your feedback.

freddair, 2020-12-14:
Install unsuccessful on Shelly 1L.

__Originally posted by @freddair in https://github.com/mongoose-os-apps/shelly-homekit/issues/323#issuecomment-744337914__

rudyemm, 2020-12-14:
Thanks @rojer for your hard work 👏

I have some feedback to report. I’ve installed the beta 1 on 4 Shelly 2.5s. I can confirm the temperature reading has dropped from by about 10-20 °C. However, I can also confirm that one of the Shellys has also crashed/rebooted as I can see its uptime is 3 hours as compared the other other Shellys that have been up for 20 hours (since I installed them).

Let me know if you’d like any more info. I’ll report back with any more findings.

pivale, 2020-12-15:
Hey @rojer , how can we validate this?

> WiFi stability improvements from Espressif SDK update and use of all-channel scan - this should make the device connect to the AP with the strongest signal instead of the first matching the SSID.

I'm on 2.7-beta1 on every device and I don't see this happening, both on 1pm and 2.5.
It seems to be connecting to the first, even if with lower RSSI. After a few reboots, it will eventually connect to the strongest one.

Setup:
- have 3 APs at home.
- separate wifi for IOT
- hidden SSID
- only 2.4ghz active on that SSID

Maybe we are not giving enough time to scan all available networks?
How can I help with this?

lucaspinelli85, 2020-12-15:
I'm testing beta. When i change switch mode (exemple in 2.5 from switch to roller shutter) don't compare the window for roller shutter but remain the windows input 1 and 2, i have to reboot device every change i save. Same thing in shelly 1 and shelly i3, when i change something need reboot to appear the dedicate windows... Is only my problem?

andyblac, 2020-12-15:
ye definatly seeing some stackover flows

[shelly.log](https://github.com/mongoose-os-apps/shelly-homekit/files/5695540/shelly.log)

lucaspinelli85, 2020-12-15:
Hi i find a bug. I'm testing shelly1 with auto off delay.
[image]
problem is with physical button. If I set auto off 1 sec, and i hold down physical button, auto off work and it go off although the button is still pressed.
If I set auto off > 1 sec, and I hold down button, auto off don't work, it stay always on although I no longer press the button. Need to press again to go off.

andyblac, 2020-12-15:
long press bypasses auto off this is correct behaviour.

andyblac, 2020-12-15:
i should document this in wiki.

lucaspinelli85, 2020-12-15:
I did not know... therefore the problem is inverse, that is, by setting 1 sec delay, prolonged pressure does not bypass the auto-off... right?

andyblac, 2020-12-15:
> I did not know... therefore the problem is inverse, that is, by setting 1 sec delay, prolonged pressure does not bypass the auto-off... right?

long press iirc requires a 1 sec hold, therefore it a matter of chicken and the egg. so in sort 1 sec delay is too short to be used with a long press.

rudyemm, 2020-12-17:
An update. I’ve got the bets on 4 devices now (all Shelly 2.5). Each has a different uptime, meaning they are rebooting themselves at various times. Also on of them only loads the top section of the UI and fails to load the rest of the page.

SilvioMigu, 2020-12-18:
Hello

Temperature Sensor AddOn for Shelly 1/1PM, do you think you can do it soon?
Thanks

rojer, 2020-12-18:
won't be 2.7, for sure.

SilvioMigu, 2020-12-18:
I'll be waiting 😊
I'm looking forward to seeing it on the Homekit
I already have it working on the shelly app.
[image]

simonfield76, 2020-12-18:
I can’t wait to see this feature enabled being fundamental for my purposes!!!

🙏🙏🙏🙏

Inviato da iPhone

[Captura de ecrã 2020-12-18 à(s) 09 48 21]<https://user-images.githubusercontent.com/67691682/102600470-c8454180-4116-11eb-90ab-ad807b817f94.png>

rudyemm, 2020-12-18:
[image]

rudyemm, 2020-12-18:
If there’s a beta 2, I’m happy to test it more @rojer

rojer, 2020-12-19:
pushed beta2 with fix for #323

rojer, 2020-12-19:
@rudyemm can you please visit http://a.b.c.d/debug/core - do you get anything there? if yes, please send to rojer@rojer.me

hfagelnour, 2020-12-19:
i am testing the beta 2.7.1 on more than 30 devices and noticed that WiFi disconnections still happens but much lesser. maybe its my infrastructure being a Unifi maybe not. Note: shelly ppl are struggling to fix this issue even on their official FW since they published 1.9.0!
just thought to run this by you.

andyblac, 2020-12-19:
> i am testing the beta 2.7.1 on more than 30 devices and noticed that WiFi disconnections still happens but much lesser. maybe its my infrastructure being a Unifi maybe not. Note: shelly ppl are [...]

my current setup is Unifi and i have no problems, but i do have AP's all over house. i have found that if RSSI reaches -80dB you will start to see disconects, this is true on Stock fw also, so i do not think it is a fw on this fw but more likely a chipset or core library issue.

i am in the progress of moving to Peplink hardware, but atm there is a firmware issue on the peplink AP's, that once you try to connect the device to HomeKit, the devcie fails to respond (no pings, no http access), even though the router shows it as connect and having been served a IP from the DHCP pool and continues to be served one. As of yet Peplink have not found the issue, but if i revert to their older fw it works fine.

hfagelnour, 2020-12-19:
i too have tons of APs at hone (18 HDs) .... disconnections is driving me crazy .... do u think having a minimum RSSI enforced fixes the matter?

andyblac, 2020-12-19:
> i too have tons of APs at hone (18 HDs) .... disconnections is driving me crazy .... do u think having a minimum RSSI enforced fixes the matter?

yes def, i control all my AP's using that, and also control the power level, remember, too strong signal is just as bad as too weak, ie try not going below -20dB that could case issues too. most of my AP's are on low power. So it only serves that one room.

hfagelnour, 2020-12-19:
@andyblac thx for the advise .... i will try it hoping it fixes things up. How does this affect roaming between rooms though. my house is heavily built with concrete and bricks

andyblac, 2020-12-19:
> @andyblac thx for the advise .... i will try it hoping it fixes things up. How does this affect roaming between rooms though. my house is heavily built with concrete and bricks

anther tip, try to only use 3 channels 1 / 7 / 11  or if in UK 3 / 8 / 13. these are the only chnnels that do not overlap

hfagelnour, 2020-12-19:
perfect.... will do that as well. i was leaving the WiFi manage things for me and apparently it is not mature enough

andyblac, 2020-12-19:
> perfect.... will do that as well. i was leaving the WiFi manage things for me and apparently it is not mature enough

make sure you have this turned off, it's located in Site, if thats on, turn it off, that will cause you nothing but trouble.

[image]

hfagelnour, 2020-12-19:
got it... many thx again

andyblac, 2020-12-19:
> got it... many thx again

np, repoort back how you get on, if you still have issue, we can try other things.

rojer, 2020-12-22:
ok, i pushed 2.7.0, thanks everyone for testing.

closed 2020-12-22 by rojer