mongoose-os-apps/shelly-homekit #331: Request: forcing Shelly to reconnect after WiFi loss

partytrik, 2020-12-12:
Hi @rojer, thank you so much for helping us all with your wonderful and speedy development!!!

Since FW 2.4 all (12) shelly 1s work very stable, but if my ap or a repeater goes down for a moment and they lose their WiFi connection, they never! reconnect until I briefly pull the electricity.

Could you implement that they automatically reconnect after a while or give us a switching pattern to force a reconnect without flipping the fuse?

This behavior (now on 2.6) is 100% identical with all shellys in these cases.

Thank you!!!

Patrik

• all Shelly 1 on fixed IPs
• 5GHz network has different name (caused issues before)
• I don’t have to reset the shellys, just pull electricity
• if I don’t want to open the wall, I need to flip fuse, but that also kills WiFi repeaters and then other shellies disconnect

rojer, 2020-12-13:
shelly does reconnect after connection loss, if it doesn't it's a bug.
to that end, i hope that 2.7.0 improves wifi connection stability, please give it a go and help testing - https://github.com/mongoose-os-apps/shelly-homekit/issues/330

partytrik, 2020-12-13:
Sure will check it out and get back to you. Thank you

partytrik, 2020-12-14:
Dear Deomid,

I have put 2.7 on all Shellies and tested with the wifi:

1) All reconnected well, fantastic.
2) While my main Router's and Repeaters’ WiFi where down, some devices managed to connect to a powerline repeater’s WiFi far away. These Shellies then sticked to that bad connection, and didn’t reconnect to the 5 times stronger WiFis, when the main Router and Repeaters came back on.
3) When I rebooted those devices that were connected to the too distant AP, they all connected correctly to the one with the best signal strength.

I have been observing this for some 40min before rebooting. Maybe you already have some sort of back up mechanism in place. But what could make sense is to have them frequently look for a stronger signal in those cases when their signal is very bad?

Thank you
Patrik

rojer, 2020-12-14:
it's true that shelly won't roam after initial connection. sdk doesn't do it, we would have to implement scanning ourselves.
perhaps, if RSSI is worse than some threshold (e.g. -70), do periodic scans and reconnect.
it's possible, but should be a separate feature request for the mongoose os wifi library - https://github.com/mongoose-os-libs/wifi

meanwhile, you confirm that 2.7 behaves better than 2.6, is that correct?

partytrik, 2020-12-17:
yes, they do reconnect now. That’s great.

OK, will open a new request then. I assume it’s a very common situation that one or more, but not all APs/repeaters go offline.

Thank you

closed 2020-12-17 by partytrik

partytrik, 2020-12-27:
Dear Deomid,

I do still have the same problem with 2.7.0. It occurs, when I do have to restart the main router. The shellies then connect to repeaters further away. Some I can just access over the web interface, reboot them and they will then connect to the main router again. But others do have a connection (so that they don’t look for one anymore), but one too bad to work in homekit or to access them over the web interface (I can see them in the router’s web interface though, including their connection strength).

The strange bit is that I found one example still working with RSSI -93, while another one was not accessible anymore at -74dbm.

Therefore, assuming that others with repeaters will experience the same kind of issues, it would make the world a lot easier, if they would frequently check for a better connection, if they have a bad one or if there was a code to switch them physically to force them to reconnect.

Since you install them in the walls, the only way to pull the electricity, are the circuit breakers. And the ones mounted in the same circuit as a repeater will always reconnect to another repeater further away, never to the repeater right next by, because the repeater will also be shut down by pulling the fuse and will afterwards need more time to start up than the shelly.

Best Patrik

Am 14.12.2020 um 15:11 schrieb Patrik Klapp <pk@persiehl.com<mailto:pk@persiehl.com>>:

Dear Deomid,

I have put 2.7 on all Shellies and tested with the wifi:

1) All reconnected well, fantastic.
2) While my main Router's and Repeaters’ WiFi where down, some devices managed to connect to a powerline repeater’s WiFi far away. These Shellies then sticked to that bad connection, and didn’t reconnect to the 5 times stronger WiFis, when the main Router and Repeaters came back on.
3) When I rebooted those devices that were connected to the too distant AP, they all connected correctly to the one with the best signal strength.

I have been observing this for some 40min before rebooting. Maybe you already have some sort of back up mechanism in place. But what could make sense is to have them frequently look for a stronger signal in those cases when their signal is very bad?

Thank you
Patrik

[Persiehl 1849 Logo]
Patrik Klapp

        Persiehl 1849 GmbH
Harvestehuder Weg 5
20148 Hamburg

+49 40 22708-100
p.klapp@persiehl.com

persiehl.com<https://www.persiehl.com>
Legal →<https://www.persiehl.com/index.php/impressum.html>

rojer, 2021-01-06:
please test 2.7.1, i fixed a whole bunch of bugs that could have affected stability.

partytrik, 2021-01-06:
Amazing. Thank you. Will do

[Persiehl 1849 Logo]
Patrik Klapp

        Persiehl 1849 GmbH
Harvestehuder Weg 5
20148 Hamburg

+49 40 22708-100
p.klapp@persiehl.com

persiehl.com<https://www.persiehl.com/>
Legal →<https://www.persiehl.com/index.php/impressum.html>