mongoose-os-apps/shelly-homekit #333: Temperature Shelly 2.5 very hot

lucaspinelli85, 2020-12-12:
Hi my shelly2.5 are very hot, is it normal?
[image]
[image]
[image]
Re ad this:
https://github.com/arendst/Tasmota/issues/7991
And this:
https://savjee.be/2020/11/shelly25-esphome-potential-fire-hazard-fix/?fbclid=IwAR30ZrNEQD27HYNRr8j3UWdjSdV7l4_J7JXXhfaQ4X8SpcoePUNy5mmzamg

rojer, 2020-12-13:
in upcoming 2.7 release an updated NTC conversion curve is used - please try 2.7.0 pre release and let me know - https://github.com/mongoose-os-apps/shelly-homekit/issues/330

you should see lower temperature readings

rojer, 2020-12-13:
arendst/Tasmota#7991 > Due to the gpio init change the IRQ gpio was never correctly initialized resulting in an output signal from the esp8266 which would be shorted to gnd by any ade irq resulting in heat and possibly hardware failure over time.

does not apply to this firmware, we don't use gpio16 so it remains configured as an input (default reset state).

closed 2020-12-13 by rojer