mongoose-os-apps/shelly-homekit #337: Add support for fan with speeds using Shelly 2.5

MichaelVdheeren, 2020-12-13:
Can we get a firmware version that allows the Shelly 2.5 to be used as a fan within Homekit with each of the inputs being mapped to a fan speed? No input would mean fan is low, input 1 means fan is medium, input 2 means fan is high.

freddair, 2020-12-14:
My bath fan has 2 speeds which correspond to 2 fan inputs / Shelly outputs: 

So for me Shelly 2.5 would need to interlock output 1 and 2.

off: both outputs off
low: output 1 on, output 2 off
high: Output 1 off, output 2 on

When both outputs are on, the fan runs very low, but I am not sure if it's safe to use as that's not specified in the manual.

andyblac, 2020-12-14:
why not just set homekit type to fan in homekit settings ?

freddair, 2020-12-14:
That's what I currently do using an Aqara double rocker switch. The problem is that it just appears as 2 separate Fan switches and not as fan speed steps, and it does not allow real interlocking of the outputs. I simulate it with scenes but it's not optimal as relays don't switch simultaneously (one after another).
[image]

MichaelVdheeren, 2020-12-14:
Correct, just changing from switch to a fan is an icon. This means I get two elements in the Home App and not a single fan with two speeds that I can set.

rojer, 2020-12-14:
HAP fan service has speed control as a %, from 0 to 100, but i guess we could have 0-30-60-100 fixed points, something like that.
i am not familiar with fans, can you explain what kind of fan are you using? what are the connections?
how common is this type of fan?

freddair, 2020-12-14:
Here's mine, built into the bathroom ceiling to exhaust steamy and/or stinky air ;-)

https://www.vortice.ltd.uk/shop/residential-ventilation/quadro/quadro-micro-100-i-es/

[image]

MichaelVdheeren, 2020-12-14:
I have exactly the same as above. Very common system in Belgium used for ventilation of the home

jsiegenthaler, 2020-12-14:
High speed is obviously 100%
Low speed could be user-configurable, or locked at 50%
The core question would be: how would homekit react to the speed control slider: would it "lock" to 50% when between 25% and 75%, and lock to 0% or 100% when outside of the 25-75% range?
Also, anyone wiring this up must be sure not to wire Live to both terminals at the same time. Some fans are simple and use a series resistor to slow down the speed, then it wouldn't matter. But some fans might have different coils or taps on the coils to change the motor speed. Then it may very much matter how it is switched.

andyblac, 2020-12-14:
my bathroom fan is a constant presure fan with boost. ie it is contantly on, and when you flip the switch it goes into boost mode. it also has it own onboard moisture sensor that spin up the fan as needed depending on humidity.

freddair, 2020-12-14:
Yeah in Germany it's also mandatory in any bathroom or kitchen at least if it has no window.

rojer, 2020-12-14:
> The core question would be: how would homekit react to the speed control slider: would it "lock" to 50% when between 25% and 75%, and lock to 0% or 100% when outside of the 25-75% range?

1 to 50 would round up to 50, 51 to 100 round up to 100..
the way it works with homekit is, home controller (aka home app) tell you to set it to something like 41, you go - wure enough, ok, then report "set to 50", and it'll update the display. at least that's the way it works with roller shutter positioning.
it tells you "set foo X", you nod, then say "foo is now Y", and it's fine :)

> Also, anyone wiring this up must be sure not to wire Live to both terminals at the same time.

sure, that can be arranged.

should also support simple on/off for singe output deices (i.e.0 and 1-100 rounding up to 100), shelly 2 and to have additional position at 50.

MichaelVdheeren, 2020-12-14:
Maybe this can help: https://github.com/adri/homebridge-itho-cve-eco-rft

jsiegenthaler, 2020-12-14:
>  1 to 50 would round up to 50, 51 to 100 round up to 100..

Wouldn't that would be quite hard to control, as you have to deliberately
get the slider under 1% to switch off the fan?
That's why I suggested 25%, it gives more of a "snap to the desired
location" behaviour.
You could do 10% & 90% instead of 25% and 75%.
But 1% and 99% is a bit extreme....

Anyway, theoretical for me, as I don't use a shelly dimmer to control my
fan ;)

timoschilling, 2020-12-15:
The HAP has a step value setting for that.
So we can say it has step value of 50. Which mean the HomeApp only allows 0%, 50% and 100%, then there is no need to round the value.

Here is an Golang example for that https://github.com/brutella/hc/blob/master/characteristic/rotation_speed.go#L16

rojer, 2020-12-15:
step value has to be exactly as specified in the standard, i.e. 1, or controller won't even add the device.

timoschilling, 2020-12-15:
If have tested it with the Simulator and it works.
[image]
[image]

timoschilling, 2020-12-15:
Here is how it looks in the HomeApp
[image]
Each time the slider jumps up or down, I released the finger at 25% or 75%.

MichaelVdheeren, 2021-01-01:
I managed to create this in homebridge-shelly. Maybe it can help you how the logic works.

https://github.com/MichaelVdheeren/homebridge-shelly/pull/1

freddair, 2021-04-03:
Any news on this?

zsolt-dev, 2021-04-11:
I would too love to skip homebridge and have this functionality directly.

My use case is a ventilation system that is controlled with 3 wires:
1 hot wire - low speed
2 hot wires - medium speed
3 hot wires - hi speed

Ventilation should never be turned off, so the first wire will be hardwired and rest of the two would get connected to shelly 2.5.

It is a bummer that I do not speak C++, so I cannot add this myself.

macintosh-HD, 2022-08-30:
Any news on this topic? I live in Germany and also have a fan with two speed settings in my bathroom. I would love to have native support for this!

rojer, 2022-08-30:
nothing so far

woodsbox, 2022-09-15:
This would be great to have!

thirsty-fatman, 2022-10-10:
I have a similar question about this post but fan setup is slightly different.

I been apprehensive about adoption of AC ceiling fans into HomeKit as our fan is controlled by Shelly 2.5 and two capacitors on two outputs. Details below:

CBB61 1.5uF+2.5uF 3 Wires AC 250V 50/60Hz Capacitor For Ceiling Fan

State   Relay0   Relay1
Off      Off      Off
Low      On       Off
Med      Off      On
Hi       On       On

Right now, the two relays appear for the fan. Relay 1 is called fan low and relay 2 is called fan medium. There is no high relay or switch but when both low and medium relays are on, then fan is running on high speed.

Is there a way that I can have one control appear in HomeKit for both relays called fan with the settings above for low, medium and high in slider? I can estimate percentages for the slider.

Many thanks in advance for your insight.

Moderators please advise if posted incorrectly or cannot be done.

thirsty-fatman, 2022-12-12:
Is my post relevant in this section or should I move my comment to a new/different thread?

PhaseAngle, 2022-12-20:
For all 3 speed capacitor fan controllers I have installed the high speed applies power direct to the fan not put the capacitors in parallel.
You need 3 relay outputs.
[image]

thirsty-fatman, 2023-01-21:
Two of the wires are input and one of the wires is output.

Electricity comes from common into the switch.

Two wires go to the capacitor with the two capacitance values. They are both switched hence the two lines in each of the circuits.

The four way rotary knob (switch) controls whether no capacitance (off,) lower rated capacitance (low,) higher rated capacitance (medium) or both capacitances (high) flow into the load (output) and then onto the fan. The fan will then spin the blades at correct speed.

I am not an electrician but based on my limited knowledge in electronics that's how I understand it.

basroovers, 2024-08-27:
This would be awesome to have. I run Homebridge just for this functionality, it would be great to have this natively.

PhaseAngle, 2024-08-27:
No that’s a 4 position switch common to off or position 1,2 or 3.