mongoose-os-apps/shelly-homekit #339: Question

hfagelnour, 2020-12-14:
I am sorry if this has been asked before but I ran a search and nothing came up.
Does this firmware have any kind of overpower protection? even if it is the device specific max?

hfagelnour, 2020-12-15:
anyone?

rojer, 2020-12-15:
no, currently we do not enforce power limits. we should, though.

hfagelnour, 2020-12-15:
Many thanks for the feedback, I believe and if it is an easy task to do is to at least enforce the maximum each device can handle just to prevent overloads and fire hazards.

rojer, 2020-12-16:
yes, we should deifinitely have this.

jorgefrazaocosta, 2021-01-09:
Hello guys,

I'm using your firmware, and one of the features provided by the original firmware, it's the "Custom Max Power Protection". By default the Shelly app shows that it should be limited to 1840W. 

Do you think this value should be applied here or we should expose it via the web interface to be configured?

rojer, 2021-01-09:
i think it should default to 1840 and user should be allowed to set value less than that, but not more.

Idomo, 2022-12-05:
@rojer 
> i think it should default to 1840 and user should be allowed to set value less than that, but not more.

Are you talking about the self consumption or the power supply for the connected device?
If it's the self consumption, how much the Shelly itself can handle without damaging anything? (before flashing this firmware I saw that the default value was 3500w, maybe it has been changed since you've checked)
If it's the power supply so it may be much more then that, for example a water heater (boiler) may take up to 3000-3500w