mongoose-os-apps/shelly-homekit #340: Temperature add-on Shelly1

Tins612, 2020-12-14:
Hello, are you planning to implement the temperature add-on for the Shelly1? I would like to switch the Shelly on when a certain temperature is reached. However, not via the Shelly app but via Homekit. Can someone help me there?

greeting

simonfield76, 2020-12-14:
I need the same as I would like to place it inside my server rack to use it as a connected thermostat, specially for the summer period!

Inviato da iPhone

svn2208, 2020-12-14:
You mean the internal temperatur sensor? This most probably won't work  (=> #265)

andyblac, 2020-12-14:
duplicate closing https://github.com/mongoose-os-apps/shelly-homekit/issues/186

closed 2020-12-14 by andyblac

Tins612, 2020-12-14:
no, I mean the external module where a temperature sensor can be terminated.

https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312

andyblac, 2020-12-14:
@Tins612 please add your support here https://github.com/mongoose-os-apps/shelly-homekit/issues/186