mongoose-os-apps/shelly-homekit #342: Led disable to current savings/lowering of temperature...

lucaspinelli85, 2020-12-15:
Yes, it is very little, perhaps irrelevant, but in original firmware there is an option to disable the led status.
Is it simple to integrate it in webui?
The devive is locked up, not visible from the outside, it is useless to have the led on which is not needed...

rojer, 2020-12-15:
the default is to have LED off if device is connected to wifi and paired with a HAP controller, so it should be off for most of the time
is that not the case?

lucaspinelli85, 2020-12-15:
You are right, sorry!

lucaspinelli85, 2020-12-18:
Not in Shelly PlugS, blue led blinking, red led on when power on...

rojer, 2020-12-18:
ok, you're right. could be done, but it's so minor i don't know if it even deserves an option.
the amount of current drawn by the LED is miniscule. there may be an argument about light pollution but... meh, i don't know.
i'll let this one simmer and gather support. if you are reading this and want this to happen, please speak up.

lucaspinelli85, 2020-12-28:
it's just the hassle of always seeing the LED lit in the plug even when it is not active ...

pivale, 2020-12-29:
It might be a problem for people with one in a bedroom during the night.
Some people have trouble sleeping with the faintest light...

chstoll, 2021-01-12:
It would be nice to get an option to disable the red LED of the PlugS. In some of my scenes the red LED is just a little disturbing, So +1 ...

closed 2021-01-26 by rojer