mongoose-os-apps/shelly-homekit #345: HomeKit + Google Home

aiFigueiredo, 2020-12-15:
Is it possible to add a device with this firmware to be controlled by the Home App in Android?

I can control my shelly 1 via Home App in iOS but the native Android one does not find the device. Also, the Shelly cloud app can not find it even when I use the shelly IP address.

Apologies if this has been answered already, but I tried searching for similar issues on the repo but could not find one similar to this.

andyblac, 2020-12-15:
google home, Alexa, or shelly app are not supported in this firmware, that could change in future, as of not it is not possible, sorry.

aiFigueiredo, 2020-12-15:
@andyblac, thank you for the response. Do you know if there is any way to have both iOS and Android to control a Shelly device?

andyblac, 2020-12-15:
only with stock firmware and use homebridge shelly plugin.

aiFigueiredo, 2020-12-15:
Thank you for the information. This issue can then be closed

closed 2020-12-15 by aiFigueiredo