mongoose-os-apps/shelly-homekit #346: Feature Request:  Shelly 1L - Support "Edge" Input Mode for Switch 2

bbfelts, 2020-12-15:
In order to use the Shelly 1L with a 3-way switch (2 switches) setup as described in the link below, I need both Switch 1 and Switch 2 to support "Edge" Input Mode.  Switch 1 currently supports "Edge" Input Mode, but there doesn't appear to be an option for "Edge" on Switch 2.  I'm assuming others trying to use the 1L for a 3-way switch setup without a neutral will also run into this issue.

https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/3206851806080828/

rojer, 2020-12-15:
hm, ok. i think we need both inputs for shelly 1L after all.
currently second input is not used and is exported as a stateless switch. as a workaround, you can add HK automations to it to control the same switch - kind of silly, but should do for now.

bbfelts, 2020-12-15:
I'm not sure I understand - If I look at the Shelly HomeKit UI on the device I see a Switch 1 and a Switch 2 - switch 2 doesn't have the "Edge" option.  I'm fine (and would actually prefer) that the HomeKit side I only see 1 switch (since there's really only one device to switch).

peer69, 2020-12-19:
Same problem here. For the hallway I’ve fallen back to stock firmware for now. Would be great to use it with this firmware.

mellsipp, 2020-12-21:
Same problem

cde-shelly, 2020-12-29:
Hi all same request here. Holding my installation for now. Thanks in advance!

nk-gx, 2021-01-07:
Some problem here. 2 switches support is absolutely necessary for that. More info here: https://peyanski.com/convert-existing-lights-into-smart-lights-shelly-1l/ and here: https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/3206851806080828/

nk-gx, 2021-01-17:
After 2.7.2 released and functionality of the second input has been restored, is it possible to add "Edge" support for Switch 2? Because without it, a 3-way installation with 2 switches turning on the same light bulb will never work properly.
Thank you in advance.

mjorgef, 2021-01-26:
This is still not working right? just install the last build and can't do this with two switches on the same room, only one switch works.

andyblac, 2021-01-26:
> This is still not working right? just install the last build and can't do this with two switches on the same room, only one switch works.

edge mode for shelly1L switch has not been added yet.

mjorgef, 2021-01-26:
> > This is still not working right? just install the last build and can't do this with two switches on the same room, only one switch works.
> 
> edge mode for shelly1L switch 2 has not been added yet.

Will it be on next update?

andyblac, 2021-01-26:
> Will it be on next update?

thats upto @rojer

rojer, 2021-01-26:
we'll see. it doesn't fit within the current architecture, requires some thinking.

mjorgef, 2021-01-26:
@rojer it Will be amazing if you were able to make it.
I have a couple of Shelly’s that work like that.
Kind regards! 🙏

hamlet-archer, 2021-02-08:
Same issue. Would be great if two-way switches are supported. Thanks! 👍

guarnacciaa, 2021-02-13:
I came here cause I'm failing to use this firmware on a 3-way installation and I guess I just found out why.
I guess the only option as of now is to go back to the stock firmware and see if it can be integrated through Homebridge.
I hope you guys will work on this asap as having no control over the Switch 2 is indeed going to impact a lot of implementations.
Keep up with the good work!

andyblac, 2021-02-13:
> I came here cause I'm failing to use this firmware on a 3-way installation and I guess I just found out why.
> I guess the only option as of now is to go back to the stock firmware and see if it can [...]

for the time being just setup the detached button in HomeKit as a ShortCut automation to toggle the light on / off

mjorgef, 2021-02-13:
> for the time being just setup the detached button in HomeKit as a ShortCut automation to toggle the light on / off

That’s a great way to do it, can you show the shortcut that you did?
Maybe I’m gonna try that option!

andyblac, 2021-02-13:
follow this example https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

wawoodwa, 2021-02-18:
Depending on the 3-way switch setup, the 1L should work.  The specific use case would be if the power enters the light box, and a 2 wire comes out of it to the first box with a 3 way switch, then a 3 wire that heads over to the second box.  The first box would have the Shelly 1L that would tie like a standard no neutral setup, with the 3 way switches connected to the SW1 and Sx. Here is a diagram of how I am going to attempt this.
[image]

dtrost87, 2021-02-27:
> I came here cause I'm failing to use this firmware on a 3-way installation and I guess I just found out why.
I guess the only option as of now is to go back to the stock firmware and see if it can be integrated through Homebridge.
I hope you guys will work on this asap as having no control over the Switch 2 is indeed going to impact a lot of implementations.
Keep up with the good work!

Hi there! Same situation for me, as for guarnacciaa. Totally agree, SW2 would complete the Shelly1L in its versatility. I don't know how complex the implementation of this task would be, but I'd also love to see it working.

nk-gx, 2021-02-27:
Workaround for Switch 2.

Connect Shelly 1L accordingly. Put Switch 1 to Edge mode. Create a Detached Switch 2 (Stateless Switch) like the following:

[image]

At "single press" action choose at the bottom "convert to shortcut" instead of selecting an accessory. Create something like that.

[image]

When it is off, turn it on and vice versa.

Hope it helps. Works perfectly for me.

mjorgef, 2021-02-27:
Promise ideia!
And you can ask Siri to turn that light on?
It will not show as a light right?

nk-gx, 2021-02-27:
Yes, because the first one looks like a light. See the image below. It's the Shelly 1L with the above configuration.

[image]

andyblac, 2021-02-27:
a detail instruction is shown in the wiki https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

mjorgef, 2021-04-14:
Hello, is this still possible for the future updates?
Instead of using the work around?!

rojer, 2021-08-18:
alright, i had a lightbulb moment, an idea how to make this happen in the code without too much twisting.
please try http://rojer.me/files/shelly/2.9.5-twoinputs1/ and let me know if this is working as expected for you.

mjorgef, 2021-08-18:
> alright, i had a lightbulb moment, an idea how to make this happen in the code without too much twisting.
> please try http://rojer.me/files/shelly/2.9.5-twoinputs1/ and let me know if this is working as expected for you.

Man, you are amazing. It is working great!!!

It keeps saying to update, but it is better not to do it right?

I chose edge on the input mode and it works great.

mjorgef, 2021-08-18:
i'm now using in three devices, and all working good with the edge option.

rojer, 2021-08-18:
> It keeps saying to update, but it is better not to do it right?

yeah, ignore that, it's a special build. you can update once an official release is made with this feature.

so, i guess, the question is - are there other important use cases for the second input?
adding an option to select would be kind of messy, ideally i would just leave it like this and be done with it.

rojer, 2021-08-18:
@timoschilling, @andyblac what do you guys think?

andyblac, 2021-08-18:
> @timoschilling, @andyblac what do you guys think?

as long as the 2nd input can still be used as "detached", so we can control other HomeKit devices.

mjorgef, 2021-08-18:
> yeah, ignore that, it's a special build. you can update once an official release is made with this feature.
>
> so, i guess, the question is - are there other important use cases for the second input? [...]

For me, like this is great!

rojer, 2021-08-18:
> as long as the 2nd input can still be used as "detached", so we can control other HomeKit devices.

yeah, that's the thing - there isn't a good place to add that option atm. i need to think a bit more.

rojer, 2021-08-18:
ok, here's my thinking - the double switch configuration only makes sense in edge and activation modes.
so, for 1L we add "Edge, 2 inputs" and "Activation, 2 inputs". if either is selected, the second input is absorbed into the switch accessory and works as is currently implemented in this test firmware. in all other modes a separate accessory is created for it, as before.

andyblac, 2021-08-18:
> > as long as the 2nd input can still be used as "detached", so we can control other HomeKit devices.
> 
> yeah, that's the thing - there isn't a good place to add that option atm. i need to think a bit more.

it already in current fw isn't it ?

andyblac, 2021-08-18:
> ok, here's my thinking - the double switch configuration only makes sense in edge and activation modes.
> so, for 1L we add "Edge, 2 inputs" and "Activation, 2 inputs". if either is selected, the [...]

sounds good

neme-at-work, 2021-08-18:
> ok, here's my thinking - the double switch configuration only makes sense in edge and activation modes.
> so, for 1L we add "Edge, 2 inputs" and "Activation, 2 inputs". if either is selected, the [...]

This could work, if we would use edge only for scenarios with two hardware switches where two edge inputs are useful. But actually all my Shellys are on edge switching since I am also using automation to trigger my outlets. If the input is not on edge, I would have to press my switch twice to trigger the outlet after it was triggered by an automation before. So edge switching is no reason to deactivate detached switching. 

I don’t use detached switching and would be happy with linked inputs, but someone might.

rojer, 2021-08-18:
> edit2: I tried to change some settings in the test version and now my shelly 1L is bricked.
> don’t change switch 1 from edge to detached mode :/
ouch, ok, i'll take a look.
meanwhile, use the [failsafe mode](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery#from-crash-looping-or-otherwise-misbehaving-firmware) to recover (upload 2.9.5 release fw)

timoschilling, 2021-08-18:
@rojer not able to test at the moment, but feature request is good and solution sounds good as well

rojer, 2021-08-18:
ok everyone, implemented the more advanced solution as described above - http://rojer.me/files/shelly/2.9.5-twoinputs2/
(users currently on `-twoinputs1` can use auto-update).
issue with setting detached mode for input is also fixed (sorry for the trouble @neme-at-work)

neme-at-work, 2021-08-19:
My shelly 1L is not showing a red LED anymore and doesn’t reset. I exchanged for another 1L I was having relay problems (relay getting stuck) before. The new firmware is flashed and both inputs work in edge mode. But sometimes triggering the input 1 switch has some weird effect.

Either my 1L is faulty or something is wrong with the code. Sometimes the relay is clicking more than once with no power on the Outlet. Also I had the light turn off and on again on one press of the switch.

rojer, 2021-08-19:
@neme-at-work can you capture and post the log while this is happening?

neme-at-work, 2021-08-19:
I hope pictures are okay.

[image]

[image]

[image]

rojer, 2021-08-19:
actually, a [gist](https://gist.github.com/) would be better, just copy-paste the whole thing and create  gist.

so what is the problem? what is the device doing while this log is being written?

neme-at-work, 2021-08-19:
Input 2 is activating and deactivating the relay.
Input 1 is just creating clicking sounds inside the shelly 1L.

rojer, 2021-08-19:
hm, it's strange. looks like all the activations on input1 are very brief, just 100 ms or so.
does it work in the normal edge mode?

neme-at-work, 2021-08-19:
Okay, I tested with the official 2.95 firmware and this shelly is definitely broken. It is not the code.

Since the other one is still bricked, I will buy new Shellys. Until they arrive I cannot test anymore. Sorry.

mjorgef, 2021-08-19:
Here it is working good on three Shelly 1L!! At the moment no issues.

rojer, 2021-08-20:
> Since the other one is still bricked

shouldn't be. are you sure there is no LED indication after power on? i think you should be able to enter failsafe mode and reflash it.

neme-at-work, 2021-08-21:
Yes, you are right. I had to press the button **before** turning the power on. Than it booted into failsafe mode.

bbfelts, 2021-08-21:
@rojer - this is awesome!  I have the -twoinputs2 installed on 4 ShellyL1's and it's working perfectly!  This build takes 4 out of 6 devices out of Hoobs and allows them to work directly with Homekit!  Now to find a way to get the 2 Shelly Dimmer 2's out of Hoobs...

rojer, 2021-08-23:
merged into master, released in 2.10.0-beta - please help test (#773).

closed 2021-08-23 by rojer

mjorgef, 2021-11-13:
Hello, my Shelly 1L with the two inputs option it start to flickering sometimes when it is off, I turn it on and off and it stops.
Is it something with the led lamps?! I already reboot it but it happen again.

neme-at-work, 2021-11-13:
The shelly 1L is very delicate. If you handle the connectors or the housing
too hard it might break. You should try another 1L.

mjorgef, 2021-11-13:
> The shelly 1L is very delicate. If you handle the connectors or the housing too hard it might break. You should try another 1L. mjorgef ***@***.***> schrieb am Sa. 13. Nov. 2021 um 03:18:
> […](#)
It’s been working great for a long time.
It appears to be one off the gu10 lamps, I’m gonna change them all and see.

mjorgef, 2021-11-17:
It looks like I had old GU10 lamps of halogen.
If I change them to leds, I need to use the Shelly bypass on each lamp?!
Or because the Shelly is controlling 6 lamps it doesn’t need?