mongoose-os-apps/shelly-homekit #347: Cannot flash Shelly2 dimmer

carboncrushin, 2020-12-15:
I've tried to flash my Shelly 2 dimmers with no success. I've tried flashing with the Shelly on the latest supported firmware and on the beta firmware and neither work.

For reference my two 1pm and my 2.5 all flashed fine.

rojer, 2020-12-16:
dimmer is not supported yet, see #14

closed 2020-12-16 by rojer

carboncrushin, 2020-12-16:
Sorry I was confused when i saw the push URL for Shelly 2 (http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly2.zip). I was making the jump in my head that this was for the "dimmer 2" not just the normal shelly 2. Given the popularity of the 1PM and 2.5 I've always found the 2 a pretty niche product as compared to the very useful dimmer 2.

My mistake!