mongoose-os-apps/shelly-homekit #349: Shelly 2.5 Roller Shutter mode lost calibration

johannes4576, 2020-12-19:
Hey Shelly friends,

the HomeKit FW is amazing! Lot better than over Homebridge or something else. But I have a problem with the calibration of the roller shutter. The calibration it self works fine but after 1 or 2 days the calibration lost and I must do it again. After the next days, the same again.

Can someone help me or can it be fixed?

Regards
Johannes

rojer, 2020-12-19:
by losing calibration do you mean the status changes to "not calibrated' in the ui?

johannes4576, 2020-12-19:
[image]

You can see it in the Screenshot, I have set the shutter this morning to 50% via Siri ... now in HomeKit the shutter is "closed" an the Shelly says idl 0. when I calibrated it new it says the % it's opend.

rojer, 2020-12-22:
"pos 0" means fully closed. 0 is closed, 100 is fully open.

johannes4576, 2020-12-22:
I know but in the screenshot the shutter is in 50% but it says 0%

rojer, 2020-12-22:
ok. as such, calibration data is not lost - we still have movement time and power recorded.
however, you say that positioning is incorrect.

>  I have set the shutter this morning to 50% via Siri ... now in HomeKit the shutter is "closed" an the Shelly says idl 0

"now in HomeKit the shutter is "closed" an the Shelly says idl 0" - this is correct, position 0 translates to fully closed.
however, you say that actually the curtain is at 50%. when you set it to 50% in the morning, did it position correctly?
when in this incorrect state, what happens if you set it to fully open? does it move?

juancitomadrid, 2020-12-22:
Hello!
The same thing happened to me.
In the last month it has happened to me in two different shellys.
Suddenly neither the opening nor the closing or even the physical button works. The blind stays open and nothing responds, but you can hear the shelly react every time you press the button or open or close it in the homekit.
I have already discovered that what happens is that it loses the calibration, it stays in idle pos = 0.
If you activate the calibration again, everything works again.

rojer, 2020-12-22:
ok, can you turn on debug log and post log when the device is misbehaving?

juancitomadrid, 2020-12-22:
> ok, can you turn on debug log and post log when the device is misbehaving?

sure!
Done.
If happen again... I will post here.

andyblac, 2021-01-27:
@johannes4576 still need help, or can this be closed ?

johannes4576, 2021-01-27:
> @johannes4576 still need help, or can this be closed ?

Yes, it can be closed. Thx

closed 2021-01-27 by andyblac