mongoose-os-apps/shelly-homekit #352: push message via automation

ArndtGold, 2020-12-20:
Certain HomeKit devices* such as locks and sensors can be set up to send you an alert when something unusual is detected. For example, when I connect the Shelly 1 to the doorbell, I would like to send a push message via automation:
https://support.apple.com/de-de/HT208939

Can your help with this. Thanks 

With kind regards

Arndt

jsiegenthaler, 2020-12-20:
No need to add anything to the Shelly1. I have a Shelly1 as my doorbell and the apple Home app notifications happen automatically.
But:
1. You must have a Video Doorbell accessory. I did mine with Homebridge and Camera-FFmpeg
2. You need to connect the doorbell button to the SW input of the Shelly1, but be careful of voltage! My Shelly1 is powered by 220VAC, and so I used a relay on my doorbell buzzer to isolate the doorbell button push from 220V. Maybe you can power the Shelly1 from your doorbell? Check the voltages!
3. You can, if you have it, connect a Door Opener. I have one on my apartment building front door, using the relay output of the Shelly1. The Shelly1 is configured as a Switch named "Door Opener" in my home app
4. Then you can do things like: Play a DingDong on Sonos speakers throughout the house, and automatically unlock the front door when you come home and press the doorbell button, but only enable for the first N minutes after entering your geofence, for extra security.

brauers, 2020-12-21:
@jsiegenthaler sounds pretty interesting ... especially the part with the doorbell sound on the Sonos speaker(s) ... can you share some more details how you done it? did I got you right, u r using 2 Shelly 1 to realize this setup?

cheers Stephan

jsiegenthaler, 2020-12-21:
@brauers I do it like this:
When the doorbell Button is on, run an automation. In the automation, I have just one step: "Run script over SSH"
The SSH step connects to my raspberry pi.  In the action, I use a bash command to execute a bash script on my pi. But you can do everything in a curl command. I have a mp3 sound track of a doorbell ringing, and I have that stored as a preset for the Sonos http command. So I just execute a pause all, play the preset, then resume all

spam667, 2020-12-21:
Hello guys, there is also an app called "PushCut" that provides you the ability to send push notifications when a Shortcut was executed.

Problem is that shortcuts launched on Home automations are like a "smaller" version of normal shortcuts so you can't send a notification from these automations. PushCut uses a tool that is able at home automation shortcuts that is "Get the content of an URL". Using this action, the app call to its servers and send the notification using the app. It is not pretty but it could be useful if you are looking for this feature.

Also, PushCut is a paid app. I don't know how much. I know this procedure because we were speaking about this topic in a HomeKit forum several weeks ago. Trying to find a solution to send notifications when something happened at home. Some are native in HomeKit like sensors, etc, but some others not and people wanted to improve notifications. Here is a way to go.

BR.

brauers, 2020-12-21:
... next project is born ... thanks guys :)

andyblac, 2020-12-21:
also, keep in mind PushCut requires and device to be used as a server, this means you will need an unused iOS devcie that is always running PushCuts in the foreground. not something i wanted to do.

ArndtGold, 2020-12-21:
Hello all, 

Thank you very much for your feedback. 

With IOS 14 reading from any NFC chip is supported. My idea is to control the electric door opener (12 DC) via a Shelly-1, which in turn can be switched by a specific NFC chip. So in total 13 EUR component cost, provided you have an iPhone 11. Of course, only as long as the door is not locked and only when my iPhone has connected to my WLAN.

Actually, there is already a corresponding function in Apple's current Home app for sending push messages to all devices (i.e. Apple TV, Apple Watch, etc.). Provided the HomeKit device (Shelly) or firmware supports this feature.

Perhaps this firmware could be supplemented accordingly ?

many greetings
Arndt

https://www.abus.com/uk/Home-Security/Door-security/Electrical-door-openers/ET75
https://www.apple.com/uk/newsroom/2020/06/apple-reimagines-the-iphone-experience-with-ios-14/

spam667, 2020-12-21:
> also, keep in mind PushCut requires a device to be used as a server, this means you will need an unused iOS devcie that is always running PushCuts in the foreground. not something i wanted to do.

Didn't know that!! I just proposed is as a possible solution but as I said, I didn't try the PushCut app...

ajkblue, 2020-12-22:
If you're looking for a doorbell system specifically, I found [this project on Gitlab](https://gitlab.com/ca-iot/homebridge-shelly-doorbell#readme) that seems like it would work with a regular doorbell

juancitomadrid, 2020-12-22:
> If you're looking for a doorbell system specifically, I found [this project on Gitlab](https://gitlab.com/ca-iot/homebridge-shelly-doorbell#readme) that seems like it would work with a regular doorbell

Hello!
The same thing happened to me.
In the last month it has happened to me in two different shellys.
Suddenly neither the opening nor the closing or even the physical button works. The blind stays open and nothing responds, but you can hear the shelly react every time you press the button or open or close it in the homekit.
I have already discovered that what happens is that it loses the calibration, it stays in idle pos = 0.
If you activate the calibration again, everything works again.

rojer, 2020-12-22:
@juancitomadrid your comment seems out of place... did you mean to respond to #349 ?

juancitomadrid, 2020-12-22:
> @juancitomadrid your comment seems out of place... did you mean to respond to #349 ?

OMG!!!! yes! sorry for this!

nobby-bln, 2020-12-25:
> 
> 
> also, keep in mind PushCut requires a device to be used as a server, this means you will need an unused iOS devcie that is always running PushCuts in the foreground. not something i wanted to do.

Nope, for push notifications done by pushcut you don't have to run the pushcut automation server. You can do this with "Get the content of an URL".
With a free account, you can use 3 different notifications.
I tried it and it is very usefull for me. As long as it's not possible to send notifications from homekit itself.

The pushcut automation server can be used as _"The Pushcut Automation Server enables you to automatically run your shortcuts, without confirmation or user interaction. After setting up a dedicated iOS device as a server all your shortcuts and HomeKit scenes will be available as server actions, ready to execute at any time."_
An example. I've tried a location based automation to switch my apple watch to silent mode at arriving at a specific location but this only display a notification and aks me, if I want to execute it. Not really what I would call an automation, right?
With the pushcut automation server this would be a real automation.

For pricing please go to [https://www.pushcut.io/](url)

andyblac, 2021-01-25:
closing as not really releated to this firmware.

closed 2021-01-25 by andyblac