mongoose-os-apps/shelly-homekit #353: Unable to Flash Shelly1L with Mongoose Firmware

notorioustubze, 2020-12-22:
So I have successfully updated all my Shelly1 devices however for some reason no matter how many times I try to update my Shelly 1L it remains on the default shelly firmware. Everything looks like the others I get the same message in the browser but no joy getting it to load on the shelly any help greatly appreciated.

andyblac, 2020-12-22:
to what version ?

notorioustubze, 2020-12-22:
{"status":"updating","has_update":false,"new_version":"20201205-115619/v1.9.2_1L@8f181f76","old_version":"20201202-135922/v1.9.3-rc3@50c6ab57","beta_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}

andyblac, 2020-12-22:
what version this fw ? 1L requires beta3

http://rojer.me/files/shelly/2.7.0-beta3/

notorioustubze, 2020-12-22:
Sorry, that's what ever firmware is being loaded from the main page instructions "http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip"
So do I need to just change the address to that above?

andyblac, 2020-12-22:
> Sorry, that's what ever firmware is being loaded from the main page instructions "http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip"
> So do I need to just change the address to that above?

try `http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip`

andyblac, 2020-12-22:
a bug was found after 2.6.0 was released, the beta should load.

notorioustubze, 2020-12-22:
So, I've used the new url and still no joy I get the following after pressing enter on the new url: 
{"status":"updating","has_update":false,"new_version":"","old_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}

andyblac, 2020-12-22:
> So, I've used the new url and still no joy I get the following after pressing enter on the new url:
> {"status":"updating","has_update":false,"new_version":"","old_version":"20201202-135922/v1.9.3-rc3@50c6ab57"}

it on dhcp or static ip ?

andyblac, 2020-12-22:
sure it has not loaded ? trying quitting browser and going to IP again.

notorioustubze, 2020-12-22:
> http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip

This one did the job sorry I missed this in your response. Thank you so much for the prompt response!! 👍

andyblac, 2020-12-22:
> > http://A.B.C.D/ota?url=http://rojer.me/files/shelly/2.7.0-beta3/shelly-homekit-Shelly1L.zip
> 
> This one did the job sorry I missed this in your response. Thank you so much for the prompt response!! 👍

thats ok, np, glad you got it sorted, i'll close this issue.

closed 2020-12-22 by andyblac