mongoose-os-apps/shelly-homekit #355: Similar issue here:

ivofernandespt, 2020-12-23:
Similar issue here:
I have an automation in HomeKit when window open (magnetic sensor), open blind and when window close, close the blind. It works as expected except if when it is opening or closing I close or open the window: If blind is opening and I close the window I want that blind starts closing but what happens is that blind stops and if open window again it will close... 

Yet with stock firmware + home bridge it works good: if blind is opening and I close the window it immediately starts closing...

__Originally posted by @ivofernandespt in https://github.com/mongoose-os-apps/shelly-homekit/issues/254#issuecomment-749851642__

rojer, 2020-12-23:
this is working as intended, and explained [here](https://github.com/mongoose-os-apps/shelly-homekit/blob/1041f8b331e4a892fc27d32a1fbacd60bdd06b1f/src/shelly_hap_window_covering.cpp#L344-L355).

rojer, 2020-12-23:
but i have to say, with all the side effects, i'm starting to think it may not be a good idea to try to guess and override what we are being told...

andyblac, 2021-03-03:
@rojer close issue or add to enhancement label ?

closed 2021-03-11 by github-actions[bot]