mongoose-os-apps/shelly-homekit #358: New Router / Problem to switch WiFi to new network

Mikerosoft90, 2020-12-23:
hello,

how do I get the shelly 25 into my new network? 
I no longer come to the web interface of the shelly to change the wlan settings. 
I have already reset the Shelly and this is also displayed as a WLAN network. In the official Shelly app I can not add the 25 and I always get an error message.

I hope you can help me :)

simonfield76, 2020-12-23:
Is there a way to assign a static IP address?

Inviato da iPhone

Mikerosoft90, 2020-12-23:
hello simon,

thanks for your reply.

do you mean that i should connect to the shelly (ap mode) and then assign a static ip? Would that help me?

BlancoDeMallloca, 2020-12-26:
If you are able to connect to the Shelly AP the IP for the web interface should be http://192.168.33.1/
In the web interface you should be able to enter the new WiFi Settings

andyblac, 2021-01-27:
@Mikerosoft90 still an issue ?

rojer, 2021-02-14:
closing in favor of #501
having multiple wifi configurations will allow for smooth migration of networks.

closed 2021-02-14 by rojer