mongoose-os-apps/shelly-homekit #359: Device name lost after upgrade to new 2.7.0 firmware

jsiegenthaler, 2020-12-23:
I just upgraded my Shelly1s form 2.6.1 to 2.7.0 and noticed that the device name gets lost.
The device name reverted to shelly1-xxxxx where xxx is I think the mac address

It would be very user-friendly if the device name could be preserved during a firmware update

andyblac, 2020-12-23:
kept name here ok, on all my devices.

jsiegenthaler, 2020-12-23:
Strange, lost it on both mine this morning. Observed on first one, wasn't sure, so watched second update closely and sure enough, name was lost. I am using 2 x Shelly1.

Stundenplan7b, 2020-12-23:
Hello, witch Firmeware did you habe before you update to 2.7?

Maybe 1.x?? I remember an information from rojer from the past. As he switch from 1.x to 2.x firmware this issue can happen!

rojer, 2020-12-23:
> It would be very user-friendly if the device name could be preserved during a firmware update

indeed we try our best to preserve user settings during upgrades. there were only a few intentional cases where user settings were not migrated, where doing so would be too complicated and impact deemed low.

which version you were upgrading from?

jsiegenthaler, 2020-12-23:
I had 2.6.1. Went from that to 2.7.0

Stundenplan7b, 2020-12-23:
Today i make the Update from 23 Shellys. 7 one's and 16 Shelly 2.5. Some with 2.5.0 and some with 2.6.1.

Change the Name on the Website and HomeKit?

andyblac, 2021-01-25:
was names ok updating to 2.7.3 ? or is this still an issue ?

jsiegenthaler, 2021-01-25:
Hi

Upgrading 2.7.1 to 2.7.3 last night occurred without any issues. You can
close this ticket.

closed 2021-01-25 by andyblac