mongoose-os-apps/shelly-homekit #360: Calibration never finishes on a long canopy / blind - when it takes above 90s to open

michaelnozbe, 2020-12-24:
Amazing work on this firmware! Thanks so much! However, there’s an issue when the canopy/blind is really big. The calibration never finishes.

I have 4 Shellys 2.5 installed in typical window blinds and no problem there.

But I also have 3 Shellys 2.5 installed on three of my terraces connected to the canopies covering them. Two of them work great, but one terrace is really big and the canopy takes lots of time to open and/or close. Above 90 seconds.

I’ve tried to calibrate this canopy but it never finishes. It opens. It closes. And then it opens 1% and stops. Reports back the time but the “calibration button” still shows like it’s calibrating. (Screenshot attached)

Can I do anything to send you a debug report or something to help out?
[image]

andyblac, 2021-01-27:
@pivale can you help out ?

pivale, 2021-01-28:
Not sure if I can help...

@rojer is there any kind of timeout or max run time for the calibration process?

andyblac, 2021-03-03:
i'l close this one, no reponce from OP,  maybe take a look here https://github.com/mongoose-os-apps/shelly-homekit/issues/404

closed 2021-03-03 by andyblac