mongoose-os-apps/shelly-homekit #363: Failsafe mode

lucaspinelli85, 2020-12-24:
What happen?
[image]

rojer, 2020-12-25:
this happens if you reboot with button held down. but i guess you did not, in this case?
you should be able to take the device out of this mode by rebooting again.

lucaspinelli85, 2020-12-25:
I find the device in this condition itself at morning when i wake up...
I don’t reboot it with button hold!!
It lost Wi-Fi, i find it because there was a Shelly Wi-Fi and i connect to it and go to 192.168.33.1
So, i try reboot but it go always in this condition (no button holded), i try to cut off power too.
I have to replace device with another shelly2.5 and then i discover the before Shelly burn the roller shutter motor, i think it close circuit on both O1 and O2 for long time...
I reset faisafe Shelly by his phisical button, only this case it back to work.
But now I have roller shutter motor burned :(
Is there a security in roller shutter mode to prevent both O1 and O2 clove circuit together?

rojer, 2020-12-25:
yes, the roller shutter algorithm ensures both outputs are never active at the same time

lucaspinelli85, 2020-12-28:
I hope it was just an isolated case. Now I have managed to restore everything.

closed 2020-12-28 by lucaspinelli85